ARMmbed/mbed-os #9119: MemoryPool: using hard defined mem pool size

0xc0170, 2018-12-17:
### Description

MemoryPool object contains this currently: `char _pool_mem[((sizeof(T) + 3) & ~3) * pool_sz];`

As we recently found out (reported by @embeddedteam103, see https://github.com/ARM-software/CMSIS_5/pull/479) this calculation is not portable.

`rtx_os.h` defines `osRtxMemoryPoolMemSize` that we can use instead. Add the calculation to our `mbed_os_storage_rtos` header file.

cc @kjbracey-arm @c1728p9 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

embeddedteam103, 2018-12-17:
The commit introducing this calculation says the size must be a multiple of 4 bytes.
Question: Is this part of the public API of cmsis-os, or implementation specific to RTX?

0xc0170, 2018-12-17:
@mprse Can you review?

embeddedteam103, 2018-12-17:
Not exactly, as this is part of the `RTX` api and not `cmsis`. 
What I proposed in ARM-software/CMSIS_5#479 is wrap this `RTX` macro in a macro in a public API of `cmsis` and use it in mbed.

0xc0170, 2018-12-17:
> Not exactly, as this is part of the RTX api and not cmsis.

True. I see this as 2 step. 
- First, we fix the hardcoded value here, as you need to redefine storage types anyway, so keep the C++ rtos objects not using RTX details.  (this should be quick to fix and easy to replace later by CMSIS definition).
- Second step, to discuss in CMSIS repo how CMSIS can provide abstraction for memory pool size and hopefully getting implemented

0xc0170, 2018-12-17:
@embeddedteam103 The issue 167 still needs to be addressed.  Can you add your comments there as you use different implementation. This will revive the discussion,

PR #10706 (2019-06-25): mem pool size: Fix calculation of memory pool size for portability
### Description

Replace the hardcoded value calculation of the memory pool block size
with the RTX preprocessor macro (via a shim layer).
This is in preparation for the replacement of the direct access of RTX
functionalities in Mbed OS with an access via CMSIS.

### Pull request type

    [ ] Fix
    [x] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@0xc0170 

fixes: #9119
closed 2019-06-25 by 0xc0170