ARMmbed/mbed-os #9123: Compiling: multiple definitions of '__aeabi_atexit'

fluidblue, 2018-12-17:
### Description

What target does this relate to?
LPC11U68

What toolchain (name + version) are you using?
GCC_ARM in mbed-cli v1.8.3

What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
mbed-cli v1.8.3

What is the SHA of Mbed OS (git log -n1 --oneline)?
From commandline output: 

    [mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at rev #e95d10626187
    [mbed] Unpacking library build "e95d10626187" in "/Users/max/Downloads/ARM-mbed/mbed-cli2/mbed_blinky/mbed"

Steps to reproduce:
1. Use mbed-cli to import Mbed OS 2 blinky example:
`mbed import https://mbed.org/teams/mbed/code/mbed_blinky/`
Hint: For LPC11U68, only Mbed OS 2 is supported.

2. Compiling with `mbed compile --target LPC11U68 --toolchain GCC_ARM` results in the following error:

		[mbed] Working path "/Users/user/Downloads/ARM-mbed/mbed-cli2/mbed_blinky" (program)
		Using targets from /Users/user/Downloads/ARM-mbed/mbed-cli2/mbed_blinky/mbed/e95d10626187/targets.json
		Building project mbed_blinky (LPC11U68, GCC_ARM)
		Scan: mbed_blinky
		Compile [100.0%]: main.cpp
		Link: mbed_blinky
		./mbed/e95d10626187/TARGET_LPC11U68/TOOLCHAIN_GCC_ARM/startup_LPC11U68.o: In function `__aeabi_atexit':
		/home/jenkins/build_node_4_1/workspace/bm_wrap/1707/mbed-os/targets/TARGET_NXP/TARGET_LPC11U6X/device/TOOLCHAIN_GCC_ARM/TARGET_LPC11U68/startup_LPC11U68.cpp:172: multiple definition of `__aeabi_atexit'
		./mbed/e95d10626187/TARGET_LPC11U68/TOOLCHAIN_GCC_ARM/mbed_retarget.o:/home/jenkins/build_node_4_1/workspace/bm_wrap/1707/mbed-os/platform/mbed_retarget.cpp:1283: first defined here
		collect2: error: ld returned 1 exit status
		[ERROR] ./mbed/e95d10626187/TARGET_LPC11U68/TOOLCHAIN_GCC_ARM/startup_LPC11U68.o: In function `__aeabi_atexit':
		/home/jenkins/build_node_4_1/workspace/bm_wrap/1707/mbed-os/targets/TARGET_NXP/TARGET_LPC11U6X/device/TOOLCHAIN_GCC_ARM/TARGET_LPC11U68/startup_LPC11U68.cpp:172: multiple definition of `__aeabi_atexit'
		./mbed/e95d10626187/TARGET_LPC11U68/TOOLCHAIN_GCC_ARM/mbed_retarget.o:/home/jenkins/build_node_4_1/workspace/bm_wrap/1707/mbed-os/platform/mbed_retarget.cpp:1283: first defined here
		collect2: error: ld returned 1 exit status

		[mbed] ERROR: "/Applications/MBEDCLI.app/Contents/Resources/miniconda/bin/python" returned error.
		       Code: 1
		       Path: "/Users/user/Downloads/ARM-mbed/mbed-cli2/mbed_blinky"
		       Command: "/Applications/MBEDCLI.app/Contents/Resources/miniconda/bin/python -u /Users/user/Downloads/ARM-mbed/mbed-cli2/mbed_blinky/.temp/tools/make.py -t GCC_ARM -m LPC11U68 --source . --build ./BUILD/LPC11U68/GCC_ARM"
		       Tip: You could retry the last command with "-v" flag for verbose output
		---

How to solve this problem?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2018-12-17:
This looks like a startup issue and should be fixed. Please send PR removing the atexit from startup file (Mbed OS retarget file provides exit handlers)

fluidblue, 2018-12-17:
@0xc0170 I created a pull request.
Is fixing this issue in the mbed-os repo also affecting the .o files that are downloaded for mbed OS 2 projects by mbed-cli?

0xc0170, 2019-01-02:
PR merged, this can be closed

closed 2019-01-02 by 0xc0170

fluidblue, 2019-01-10:
@0xc0170 I still can't get it to compile.

Currently, library build "e95d10626187" (2018-09-06) is used.
When issuing `mbed update`, it states this is the latest version.
When using `mbed update --latest-deps`, it updates to "3a7713b1edbc" (2018-11-08).
However this is still not an up-to-date version.

How can I obtain the latest revision that includes the fix in this issue?

0xc0170, 2019-01-15:
the next mbed 2 release should include it (you can just fork and apply the patch for now).

> When using mbed update --latest-deps, it updates to "3a7713b1edbc" (2018-11-08).

This is currently the latest release. We will release a new update with the next patch release

fluidblue, 2019-01-17:
@0xc0170 ok, thanks!

For others: Using [mbed-dev](https://os.mbed.com/users/mbed_official/code/mbed-dev/) (mbed os 2 sources) and applying the patch manually works.