ARMmbed/mbed-os #9132: MTB_UBLOX_NINA_B1 (nRF52) BLE reconnecting to iOS on power cycle

edablan, 2018-12-18:
### Description
- I have setup a custom BLE profile with 3 services. I can successfully connect and pair with the device, as well as bond. However the device does not have any external NVM memory to store the LTK during bonding. I am trying to avoid having to manually "forget the device" in iOS BLE settings to force pairing and bonding again after the first time, or continually have iOS ask the user if they want to pair to the device. I guess there are a few ways to accomplish this, but none completely take away iOS re-prompting the user to pair with the device, unless I properly store the keys in NVM during bonding. 

I have tried manually turning off LE secure connections, along with encryption being turned off. My thought was using Just Works legacy pairing with no bonding would simply connect to the device without needing to prompt or even re-prompt the user from iOS. Unfortunately that's not the case.

I tested the arm mbed-os-example for Battery service and  can connect and re-connect without being prompted on power cycle, and it appears bonding is setup for the device because iOS remembers the device in BLE settings. However, I don't see  any security manager settings being setup for the Battery example though, and not sure how the LTKs seem to be stored properly for this example on power cycle. Also, it never prompts to pair with the device in iOS, it just connects the very first time.

I guess I'm wanting the behavior seen in the BatteryService example, but if I remove all of the securitymanager settings in my custom profile, and make it look like the BatteryService example (which doesn't seem to do anything with the security manager) I can't even connect to the device. 

Any help would be appreciated on what I can do to get bonding to work. I noticd there is a flash-driver for the IAP, but that seems experimental and not sure if would work with the nRF52. Regardless bonding seems to work with the BatteryService on power cycle and never prompts to pair from iOS.

- MTB_UBLOX_NINA_B1 (nRF52)
- IAR 7.8.40
- mbed cli 1.7.2
- mbed-os-5.11.0 (6a0a86538)

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2018-12-19:
@ARMmbed/mbed-os-pan Could y'all take a look at this?

edablan, 2018-12-19:
Alright seems I have been able to get iOS to stop asking me if I want to pair with a device every time or even on the first connect. The issue stems with having leftover setSecurityRequirements() for the characteristics created inside the DeviceInformationService constructor with parameters set to
```C++
GattCharacteristic::SecurityRequirement_t::UNAUTHENTICATED
```
I had these setup for BLE HID testing a long time ago, since that has specific security requirements. If I change those to NONE (was UNAUTHENTICATED) the problem goes away. Better yet if I remove  setSecurityRequirements()  from DeviceInformationService all together, the default is NONE, and the problem goes away. 

Still no explanation on the weirdness seen in windows, but I'll keep trying to come up with something.

pan-, 2018-12-20:
@edablan Thanks for the detailed report I opened an issue to improve the user experience regards to the security manager, its initialisation and security requirements of a characteristic. 

About the Windows behaviour, it would require some investigation to understand what triggers the pairing. The docs doesn't said much either ([1](https://docs.microsoft.com/en-us/windows/uwp/devices-sensors/pair-devices), [2](https://docs.microsoft.com/en-us/windows/uwp/devices-sensors/bluetooth-dev-faq))

linlingao, 2019-04-30:
Reopen the issue to address Windows behavior.

closed 2019-10-31 by ciarmcom