ARMmbed/mbed-os #9133: Handling no RTOS is done wrong

embeddedteam103, 2018-12-18:
### Description

This is a very annoying little bug:
In certain in mbed there is conditioning on whether rtos is present or not (or any other module). The problem is that in many place the conditional to check for it is wrong (which causes a compile time error).
The check should be
```
#if MBED_CONF_RTOS_PRESENT
....
#endif
```

But in many places it is defined as `#ifdef` or `#if defined()` which give the wrong result.

The way to remove rtos or anything else optional in mbed (the way I understand it, maybe I'm wrong) is to have an `mbed_app.json` file and have there `module-name : {present: "0"}` that overrides the defaulted `"1"` value.
When we do this - the macro `MBED_CONF_RTOS_PRESENT` is defined to be zero and therefore the tests for it should be the value and not if it is defined.
I see https://github.com/ARMmbed/mbed-os/pull/8957/files which encounters a similar problem.

Some of the files affected (we used a number of regex searches in mbed 5.8.3 with some modules missing to produce the list, so it might not be complete):

- events\mbed_shared_queues.h
- events\mbed_shared_queues.cpp
- events\equeue\equeue_platform.h
- events\equeue\equeue_mbed.cpp
- platform\mbed_retarget.cpp
- platform\mbed_sdk_boot.cpp
- platform\Platformmutex.h
- platform\SingletonPtr.h
- platform\mbed_poll.cpp
- platform\mbed_wait_api_rtos.cpp

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

kjbracey-arm, 2018-12-18:
I think you can override with `present: null` - I think a `null` will override a non-null fine, and JSON `null` leads to the value not being defined.

But in general you're right, tests on `MBED_CONF` should normally be `#if` on the assumption that they will be either `true` or `false` . (`null` would lead to not defined which `#if` treats as `0`)

(JSON `true` and `false` are also now permitted rather than `0` or `1` - this wasn't the case in early versions of the config tools).

0xc0170, 2018-12-18:
> But in general you're right, tests on MBED_CONF should normally be #if on the assumption that they will be either true or false . (null would lead to not defined which #if treats as 0)

👍 

@ARMmbed/mbed-os-tools Please review . As mentioned, this touches as well 8957. This ifdef vs if confuses people. I can't spot this documented in https://os.mbed.com/docs/latest/reference/configuration.html

kjbracey-arm, 2018-12-18:
> The way to remove rtos or anything else optional in mbed (the way I understand it, maybe I'm wrong) is to have an mbed_app.json file and have there module-name : {present: "0"} that overrides the defaulted "1" value.

That isn't actually the direction it was intended to work, I believe - it was really a detection mechanism for the case where someone had removed the directory. It was for code that needs to work on both Mbed OS 2 and Mbed OS 5, rather than an Mbed OS 5 config mechanism.

But I guess it can work in the way you describe - if you undefine the value, no-one will call it, so the linker will end up excluding it.

Code which is intended to be excluded at compile time is flagged via `feature` or `component` or `target` switch gating. The RTOS is not currently officially excludable for Mbed OS 5, but a lot of people do it unofficially by having an `.mbedignore` file that skips the directory.

(Build config wasn't the intent of mbedignore - it was to ignore files totally irrelevant to mbed OS like other OS port directories, so people didn't have to restructure their upstream repos if going in as a subtree).

bridadan, 2018-12-18:
@kjbracey-arm is correct, setting the `MBED_CONF_RTOS_PRESENT` macro (or the equivalent configuration option) is not the intended way of removing the RTOS. It is possible to build Mbed OS without the RTOS by adding the `rtos` directory to your `.mbedignore` file, as was previously stated.

@0xc0170 I'll review on #8957.

embeddedteam103, 2018-12-18:
Just a comment: I think the reason this was left undetected/unchanged was the fact that adding `rtos` to `.mbedignore` is the semi-official way to disable RTOS (which avoids the problem).

0xc0170, 2018-12-19:
@embeddedteam103 Is there anything left here for building without RTOS? We will look at how to improve it and make it official way.

deepikabhavnani, 2018-12-20:
> Just a comment: I think the reason this was left undetected/unchanged was the fact that adding rtos to .mbedignore is the semi-official way to disable RTOS (which avoids the problem).

Partially because of .mbedignore and partially because few components and features do not have RTOS dependency handled fully.

bmcdonnell-ionx, 2019-01-10:
Maybe relevant:
https://github.com/janjongboom/mbed-os-example-blinky-no-rtos

0xc0170, 2019-08-14:
@embeddedteam103 How this changes with baremetal profile introduction : https://os.mbed.com/docs/mbed-os/v5.13/reference/mbed-os-bare-metal.html ?

adbridge, 2020-10-13:
As there has been no response to our question for over a year , I am closing this. If this is still an issue with Mbed 6 and / or baremetal please raise a separate issue.

0xc0170, 2020-11-02:
> @embeddedteam103 How this changes with baremetal profile introduction : https://os.mbed.com/docs/mbed-os/v5.13/reference/mbed-os-bare-metal.html ?

Even if  this is not an answer, I'll close this and a new issue should be crated.

 Please look at the work we are doing on feature-cmake (soon to be on master) with new tools. There should be `mbed-baremetal` target.

closed 2020-11-02 by 0xc0170