ARMmbed/mbed-os #9137: How to build `mbed` lib only without the project root

embeddedteam103, 2018-12-18:
### Description

I'm not sure if it's a bug, enhancement, question or all of the above :)
We are using `mbed` as one library of many that our main library depends on.
This means that the `mbed-os` directory is not sitting in our project root.

We want to use a different build system to manage our build (as we do not use the entire SDK functionality).

We have the following application structure:

```
Application
----mbed_config_files/
-------- custom_targets.json
-------- mbed_app.json
-------- .mbed
-------- .mbedignore
-------- mbed_settings.py
----Lib A
----Lib B (internal dependency)
--------Lib C (external dependency)
--------mbed-os (external dependency)
```

Note this is not necessarily reflected in directory structure (other then the `mbed_config_files` directory) as it is managed by a package manager (`conan`).
All we want is to pass the configuration files to mbed-os and build it with the `mbed-cli` tool (alone).
The linking and managing of building our source code is done by our build system, and a package manager links the whole thing together.
The build process is as follows:

1.  (Package Manager) Fetch relevant versions of all packages
2.  (Package Manager) Start Build Lib A
3.  (`CMake`) Lib A build
4.  (Package Manager) Start Build Lib B
5.  (Package Manager) Start Build Lib C
6.  (`make`) Build Lib C
7.  (Package Manager) Start Build `mbed-os`
8.  (**python hack**)  Build `mbed-os`
9.  (CMake) Build Lib B
5.  (Package Manager) Start Build Application
6.  (`CMake`) Build Application
10. (Package Manager) Link Application

One thing to notice is that if we build mbed-os in a separate way, then all of additional include directories added by the `mbed-cli` are not propagated to the library and application (which is what we want).

Right now we sort-of-hack it by creating an mbed program inside the `mbed-os` directory in `Lib B` (with `mbed new LibB\mbed-os`)
What we really want is the partial functionality of building mbed-os only with the `mbed-cli` or the `tools` directory by providing the configuration files.

We don't want to use the fancier stuff because it makes the entire system be around `mbed-os` which does not fit to our toolchain.

So the question boils down to - is there an official (and documented) way to build `mbed-os` (ONLY) with custom configurations, without hacking around the way `mbed-cli` works?

I ask this here, because I don't know if the features I look forward to already exist in the `tools` directory but are undocumented as public API, or they don't exist.

If it is a matter of exposing internals as part of the API we can and will help.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

embeddedteam103, 2019-04-29:
Explanation for closing?

40Grit, 2019-04-29:
linlingao

linlingao, 2019-04-29:
Changed to Enhancement.

closed 2019-10-31 by ciarmcom