ARMmbed/mbed-os #9138: Can't execute OS2 tests in ARM if default_toolchain is uARM

jeromecoutant, 2018-12-18:
### Description

mbed version: master (#8305 included) + #9107 

We can't execute OS2 tests in ARM if default_toolchain is uARM

ex:
$ python ./tools/build.py -m NUCLEO_F042K6 -t ARM
Build successes:
  * ARM::NUCLEO_F042K6

$ python ./tools/make.py -m NUCLEO_F042K6 -t ARM -n MBED_2
Building project stdio (NUCLEO_F042K6, ARM)
Could not compile for NUCLEO_F042K6: No Linker Script found

If I removed from targets.json file:
"default_toolchain": "uARM",
=> it's working

https://os.mbed.com/docs/v5.10/tools/adding-and-configuring-targets.html
The default_toolchain property names the toolchain that compiles code for this target in the Online Compiler.

=> so default_toolchain should not impact python scripts ?

@theotherjimmy @bridadan 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

bridadan, 2018-12-18:
Hi @jeromecoutant,

I just checked the commands you posted above, doesn't look like `make.py -n <test name>` works for any target, not just targets with `uARM` set as the `default_toolchain`. `singletest.py` continues to build them fine. Can you use that script for now?

jeromecoutant, 2018-12-18:
$ python ./tools/build.py -m NUCLEO_F030R8 -t ARM
$ python ./tools/make.py -m NUCLEO_F030R8 -t ARM -n MBED_2

This is working

bridadan, 2018-12-18:
Ah oops, thanks. Been a while since I've used these scripts. I was able to reproduce the issue.

theotherjimmy, 2019-01-02:
@bridadan None. That sounds correct.

PR #9297 (2019-01-17): uARM targets now build in correct ARM_MICRO directory
### Description

Fixes #9138.

This is for OS2 testing. Previously, targets that were configured with
the option "default_toolchain": "uARM" would still build to a
TOOLCHAIN_ARM_STD directory. This fixes these targets to build into
TOOLCHAIN_ARM_MICRO.

@theotherjimmy I'm not 100% in love with how I have to fudge the toolchain names, do you have any other ideas on how to patch this?

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking [... cut]
closed 2019-01-17 by 0xc0170