ARMmbed/mbed-os #9141: Async spif block device?

loverdeg-ep, 2018-12-18:
### Description

Looking to understand feasibility regarding implementation of a spif block device that would use async spi writes.

Any thoughts?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2018-12-18:
@ARMmbed/mbed-os-storage Thoughts on the block device part?
@ARMmbed/mbed-os-hal Thoughts on the async-spi part?

davidsaada, 2018-12-19:
This should require a different block device class comparing to `BlockDevice`, in which all APIs are blocking. We will need to implement a new "AsyncBlockDevice" base class with asynchronous flavored APIs, such as Read/Write/Erase start and either poll on actions or some sort of end action registration. It's feasible, but not with current base class.

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@loverdeg-ep It appears your question has been answered. Please let us know if you have further questions.