ARMmbed/mbed-os #9142: softfp does not work for Cortex-M33

mmahadevan108, 2018-12-18:
### Description
Cortex-M33 is using softfp floating point implementation 
https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L426

However the __FPU_USED flag is not set thereby the FPU is not enabled and crashes when executing floating point instructions.

### Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2018-12-19:
@ARMmbed/mbed-os-tools Please review

kjbracey-arm, 2018-12-19:
Is this a CMSIS bug?

For ARMC6, core_cm33.h is checking `__ARM_PCS_VFP`, but that's a Procedure Calling Standard check. It's false because softfp is selected.

I believe it should be checking `__ARM_FP` ("hardware floating-point is available"), which is the equivalent of the `__TARGET_FPU_VFP` it uses for ARMC5.

I would expect the same failure to be seen on Cortex-M7F devices with ARMC6, which are also set to softfp. But M4F are set to hardfp, so avoids this problem. But it shouldn't - all devices should be using the same ABI, so that should be softfp.

kjbracey-arm, 2018-12-19:
And I think we've got a real issue with FP ABI inconsistency on ARMC5, ARMC6 and IAR. GCC is the only one always setting softfp.

I guess the issue doesn't show up because we don't ever pass FP values in and out of precompiled binary libraries. I wonder whether we should resolve on consistent use of softfp, or let it vary based on hardware, and just say no FP passing to binary libraries.

I assume the linkers would warn if FP was used with a mismatch (as it isn't warning about any mismatch generally).

kjbracey-arm, 2018-12-19:
Raised #9153 about the FP ABI issue - I don't think we should ever be selecting `softfp` for ARMC6.

PR #9150 (2018-12-20): Check correct ARMC6 predefine for FP codegen

### Description

For ARMC6, CMSIS headers were checking the `__ARM_PCS_VFP`, which indicates hardfp ABI in use, when they need to check whether FP code generation is enabled. Change this to `__ARM_FP`, so it works for platforms using softfp ABI.

Change already present in CMSIS_5 repo, via commit https://github.com/ARM-software/CMSIS_5/commit/969822ae162539d50617d1e5a3634ee2fd3b60f6, but redone with local search-and-replace.

Resolves #9142

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ [... cut]
closed 2018-12-20 by 0xc0170