ARMmbed/mbed-os #9145: dataflash + littlefs fatal runtime error 0x80FF0100

geoffrey-vl, 2018-12-19:
### Description

During testing with mbed-os-example-filesystem I found DataFlash + FAT not to be working in our custom target. We have a AT45 with 64Mbit storage, but also a SDCard slot.
However following configurations are working:
SDCard + FAT
SDCard + littleFS
AT45 + littleFS

Using mbed-os 5.10 (610e35ddc6d59f153173c1e7b2748cf96d6c9bcd) and mbed-os-example-filesystem (4cf621ec05204d364a72a3a4f0bc0d54dfb8af11)
Also repeated the same test on mbed-os 5.9.2 and older version of mbed-os-example-filesystem.

```
--- Mbed OS filesystem example ---
Mounting the filesystem... Fail :(
No filesystem found, formatting... Fail :(

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x800089F
Error Value: 0x0
Current Thread: Id: 0x20005924 Entry: 0x8007843 StackSize: 0x1000 StackMem: 0x20005968 SP: 0x200068B8
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0100
-- MbedOS Error Info --
error: I/O error (-5)
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

dhwalters423, 2018-12-20:
@geoffrey-vl can you please edit the title to say dataflash + fat?

cmonr, 2018-12-20:
@ARMmbed/mbed-os-storage

davidsaada, 2018-12-23:
First of all, I think that FAT file system is out of its natural neighborhood in this configuration. FAT FS is tailored to work with disks/SDs and not with NOR flash components. Dataflash components, which may produce some awkward erase sizes, may worsen the situation. That said, run time errors, like the one you see, are of course not acceptable.
I've tried reproducing the problem on my board (K64F with a Dataflash shield), but couldn't as FAT FS asserts that my storage size is too small (guess this is yet another FAT FS limitation). Would like to ask you to add the following code to the start of main.cpp and tell me the results, so I may be able to simulate your problem:
```
    bd->init();
    printf("BD size: %lld\n", bd->size());
    printf("BD read size: %lld\n", bd->get_read_size());
    printf("BD program size: %lld\n", bd->get_program_size());
    printf("BD erase size: %lld\n", bd->get_erase_size());
    bd->deinit();
```

davidsaada, 2018-12-23:
FYI @dannybenor

geoffrey-vl, 2019-01-02:
```
Board ready
--- Mbed OS filesystem example ---
BD size: 8650752
BD read size: 1
BD program size: 1
BD erase size: 2112
Mounting the filesystem... Fail :(
No filesystem found, formatting... Fail :(

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x80008EB
Error Value: 0x0
Current Thread: Id: 0x20005924 Entry: 0x800789B StackSize: 0x1000 StackMem: 0x20005968 SP: 0x200068B8
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0100
-- MbedOS Error Info --
error: I/O error (-5)

```

I'm a bit worried though about that BD erase size, should it be 512?

davidsaada, 2019-01-02:
Thanks @geoffrey-vl. Will try to reproduce your problem now.
As for the erase size - it's probably OK. Dataflash components have awkward erase sizes.

davidsaada, 2019-01-03:
@geoffrey-vl OK, managed to reproduce the problem (simulating your Dataflash component). 
Problem is indeed related to the Dataflash sector size - FAT FS won't accept any sector size that's not a power of 2. See the code [here](https://github.com/ARMmbed/mbed-os/blob/f9e840efbd2d8526fe2bb388bf02bc04ca268d48/features/storage/filesystem/fat/ChaN/ff.cpp#L5576). As such, it practically won't support Dataflash with its awkward sector sizes.
Nothing much to do here other than maybe adding a more informative error message in FatFS.

geoffrey-vl, 2019-01-03:
Hmmm, though I remember it was working in the past using mbed-os 5.7

davidsaada, 2019-01-03:
> Hmmm, though I remember it was working in the past using mbed-os 5.7

Very easy to check (just by changing the hash in mbed-os.lib and running `mbed deploy`). Would surprise me though if it works - Don't think this code has ever changed (we just moved it to a new directory).

geoffrey-vl, 2019-01-07:
I can confirm it was working in the past.
mbed-os 5.7.5: 569159b784f70feaa32ce226aaca896fb83452f7
sd-driver: f4ab55df7768cfcb049b522bebd30218ee729c81
dataflash-driver: efcb8a1a5c11225028b175b64af4149532e3ef97

```
######################################
Board ready
######################################
#    Mbed OS filesystem benchmark    #
######################################
************************************
*       Benchmarking SD-card       *
************************************
TAGS: 500
dataflash size: 7826571264 bytes
dataflash read size: 512 bytes
dataflash program size: 512 bytes
dataflash erase size: 512 bytes
************************************
*          Benchmarking            *
************************************
[DEBUG] FOLDER /fs:
    .
    ..
    tags.db
Total 3 files in /fs directory.
[DEBUG] DELETE TAGS FILE
    .
    ..
Total 2 files in /fs directory.
[DEBUG] CREATING TAGS FILE
[DEBUG] FILE CONTENTS
[DEBUG] FOLDER /fs:
    .
    ..
    tags.db
Total 3 files in /fs directory.
...
************************************
*        Benchmarking AT45         *
************************************
TAGS: 500
dataflash size: 8650752 bytes
dataflash read size: 1 bytes
dataflash program size: 1 bytes
dataflash erase size: 2112 bytes
************************************
*          Benchmarking            *
************************************
[DEBUG] FOLDER /fs:
    .
    ..
    tags.db
Total 3 files in /fs directory.
[DEBUG] DELETE TAGS FILE
    .
    ..
Total 2 files in /fs directory.
[DEBUG] CREATING TAGS FILE
[DEBUG] FILE CONTENTS
[DEBUG] FOLDER /fs:
    .
    ..
    tags.db
Total 3 files in /fs directory.
...
######################################
# Mbed OS filesystem benchmark done! #
######################################
```

davidsaada, 2019-01-07:
Thanks @geoffrey-vl. It appears from your prints that the culprit is the dataflash driver, reporting two different erase sizes on the same component. Will investigate.

geoffrey-vl, 2019-01-07:
But they're 2 different block-devices so that's OK? Erase size remains the same per block-device when switching mbed-os versions

davidsaada, 2019-01-07:
Sorry I overlooked that in your prints - thought the first erase size belonged to dataflash. Will look at the changes in the FAT FS code then.

geoffrey-vl, 2019-01-07:
Either that, or make sure it properly documented. For us it is not a requirement to work just something handy to include in testing and comparing storage options.

chrissnow, 2019-05-31:
This should work if you use binary multiple page sizes,
It will need a format but add this into your target_overrides in mbed_app.json
"dataflash.binary-size": "1"

0xc0170, 2019-08-14:
@geoffrey-vl Was this fixed or is there still any action item ?

geoffrey-vl, 2019-08-14:
I haven't tested this yet, but I'm on another project at this moment so for now its no longer an action item. I'll close this for now.

closed 2019-08-14 by geoffrey-vl