ARMmbed/mbed-os #9147: DISCO_L475VG_IOT01A does not pass Mbed OS networking tests

SeppoTakalo, 2018-12-19:
### Description

**DISCO_L475VG_IOT01A** board is not able to pass all Mbed OS networking tests as according to [Network Socket test plan](https://github.com/ARMmbed/mbed-os/blob/master/TESTS/netsocket/README.md)

**Results**

```
+-----------------------------+---------------------+---------------------------------+---------+--------------------+-------------+
| target                      | platform_name       | test suite                      | result  | elapsed_time (sec) | copy_method |
+-----------------------------+---------------------+---------------------------------+---------+--------------------+-------------+
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | TIMEOUT | 140.55             | default     |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | FAIL    | 142.44             | default     |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-udp     | FAIL    | 91.3               | default     |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-network-interface | TIMEOUT | 509.85             | default     |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-network-wifi      | OK      | 78.11              | default     |
+-----------------------------+---------------------+---------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 2 FAIL / 1 OK / 2 TIMEOUT
mbedgt: test case report:
+-----------------------------+---------------------+---------------------------------+---------------------------------------+--------+--------+---------+--------------------+
| target                      | platform_name       | test suite                      | test case                             | passed | failed | result  | elapsed_time (sec) |
+-----------------------------+---------------------+---------------------------------+---------------------------------------+--------+--------+---------+--------------------+
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS                      | 0      | 0      | ERROR   | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_CACHE                | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_CANCEL               | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_EXTERNAL_EVENT_QUEUE | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_INVALID_HOST         | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC  | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_SIMULTANEOUS         | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_SIMULTANEOUS_CACHE   | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_SIMULTANEOUS_REPEAT  | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | ASYNCHRONOUS_DNS_TIMEOUTS             | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | SYNCHRONOUS_DNS                       | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | SYNCHRONOUS_DNS_INVALID               | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-dns     | SYNCHRONOUS_DNS_MULTIPLE              | 0      | 0      | SKIPPED | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_BIND_ADDRESS                | 0      | 1      | FAIL    | 0.11               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_BIND_ADDRESS_INVALID        | 0      | 1      | FAIL    | 0.11               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_BIND_ADDRESS_NULL           | 0      | 1      | FAIL    | 0.1                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_BIND_ADDRESS_PORT           | 0      | 1      | FAIL    | 0.1                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_BIND_PORT                   | 0      | 1      | FAIL    | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_BIND_PORT_FAIL              | 0      | 2      | FAIL    | 0.1                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_BIND_UNOPENED               | 1      | 0      | OK      | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_BIND_WRONG_TYPE             | 0      | 1      | FAIL    | 0.11               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_CONNECT_INVALID             | 1      | 0      | OK      | 0.21               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_ECHOTEST                    | 1      | 0      | OK      | 2.6                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_ECHOTEST_BURST              | 1      | 0      | OK      | 12.12              |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_ECHOTEST_BURST_NONBLOCK     | 1      | 0      | OK      | 11.85              |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_ECHOTEST_NONBLOCK           | 1      | 0      | OK      | 23.2               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_ENDPOINT_CLOSE              | 0      | 1      | FAIL    | 0.56               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_OPEN_CLOSE_REPEAT           | 1      | 0      | OK      | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_OPEN_DESTRUCT               | 1      | 0      | OK      | 9.99               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_OPEN_LIMIT                  | 1      | 0      | OK      | 0.22               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_OPEN_TWICE                  | 1      | 0      | OK      | 0.0                |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_RECV_100K                   | 1      | 0      | OK      | 3.38               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_RECV_100K_NONBLOCK          | 1      | 0      | OK      | 3.92               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_RECV_TIMEOUT                | 1      | 0      | OK      | 0.78               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_SEND_REPEAT                 | 1      | 0      | OK      | 6.07               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-netsocket-tcp     | TCPSOCKET_SEND_TIMEOUT                | 1      | 0      | OK      | 0.21               |
[... cut ...]
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | mbed-os-tests-network-wifi      | WIFI-SET-CREDENTIAL                   | 1      | 0      | OK      | 0.12               |
+-----------------------------+---------------------+---------------------------------+---------------------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 16 FAIL / 15 SKIPPED / 44 OK / 2 ERROR
mbedgt: completed in 962.54 sec
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

SeppoTakalo, 2018-12-19:
CC: @screamerbg

0xc0170, 2018-12-19:
cc @ARMmbed/team-st-mcd

LMESTM, 2018-12-19:
@SeppoTakalo thank you for the report.
What version of the wifi driver have you been using ? 
Does it include the latest commits that have been pushed 2 days ago ? 
cc @jeromecoutant

jeromecoutant, 2019-01-08:
Need https://github.com/ARMmbed/wifi-ism43362/issues/31

SeppoTakalo, 2019-01-30:
[image]
[image]

Now looking good!
Thank you for fixing these.

closed 2019-01-30 by SeppoTakalo