ARMmbed/mbed-os #9149: SPI working unpredictably when 2 different SPI devices on 2 different SPI buses are heavily used simultaneously.

jarlamsa, 2018-12-19:
### Description

SPI working unpredictably when 2 different SPI devices on 2 different SPI buses are heavily used simultaneously.

Current SPI implementation has a global mutex and a global owner pointer which are preventing using  different SPI buses and peripherals simultaneously. The RF drivers under mbed-os are omitting the use of the global mutex on the SPI-class because they are very high priority and the global mutex might prevent their usage. Following fix made with @kjbracey-arm fixes the issue.
https://github.com/jarlamsa/mbed-os/commit/3c23a745d988ed0fdd166d65adb9e1c90b5e6c86

The SPI is currently going through a large modification in https://github.com/ARMmbed/mbed-os/pull/8445 and will also solve the issue.

The PR is targeting a feature branch and it is not clear when it is merged to master. This issue needs to be resolved before mbed-os 5.12 is released and is considered as a blocker if it is not resolved.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

deepikabhavnani, 2018-12-28:
@jarlamsa - Feel free to add the PR of proposed solution. jarlamsa@3c23a74

jarlamsa, 2019-01-02:
@deepikabhavnani, unfortunately that is not a proposed solution, it is just a temporary workaround. @kjbracey-arm, do you have some proposition for the fix before the #8445 is merged?

PR #9469 (2019-03-01): SPI upgrade - per-peripheral mutex and GPIO-based SSEL
### Description

This commit takes some of the work done on the SPI class from #8445, and refines it, to provide the per-peripheral mutex functionality.

This also implements GPIO-based SSEL, which exposes a new `select()/deselect()` API for users to group transfers, and should work on every platform (unlike the HAL-based SSEL). This requires users to use a new constructor to avoid backwards compatibility issues.

To activate the per-peripheral mutex, the HAL must define `SPI_COUNT` and provide `spi_get_peripheral_name()`.  (In #8445 this  [... cut]
commit 5e059b7d (2019-03-01): SPI upgrade - per-peripheral mutex and GPIO-based SSEL

This commit takes some of the work done on the SPI class from #8445, and
refines it, to provide the per-peripheral mutex functionality.

This also implements GPIO-based SSEL, which exposes a new
select()/deselect() API for users to group transfers, and should work on
every platform (unlike the HAL-based SSEL). This requires users to use a
new constructor to avoid backwards compatibility issues.

To activate the per-peripheral mutex, the HAL must define SPI_COUNT and
provide spi_get_peripheral_name().  (In #8445 this is a reworked
spi_get_ [... cut]
closed 2019-03-01 by 0xc0170