ARMmbed/mbed-os #9169: Test requiring application binary not build

salil0907, 2018-12-20:
Test requiring application binary not build

-->

### Description

Is there anything I am missing?

C:\Users\ssir\Downloads\mbed-os-example-ble-master\BLE_Beacon>mbed test -m  DISCO_F429ZI  -t GCC_ARM --icetea
[mbed] Working path "C:\Users\ssir\Downloads\mbed-os-example-ble-master\BLE_Beacon" (program)
Building library mbed-build (DISCO_F429ZI, GCC_ARM)
Scan: BLE_Beacon
Building project socket_app (DISCO_F429ZI, GCC_ARM)
Scan: GCC_ARM
Scan: socket_app
Compile [ 25.0%]: cmd_ifconfig.cpp
[Error] cmd_ifconfig.cpp@27,2: #error [NOT_SUPPORTED] No network configuration found for this target.
Building project exampleapp (DISCO_F429ZI, GCC_ARM)
Scan: GCC_ARM
Scan: exampleapp
Compile [100.0%]: main.cpp
[Error] main.cpp@22,2: #error [NOT_SUPPORTED] Skipping example application.
Memory map breakdown for built projects (values in Bytes):
| name | target | toolchain | static_ram | total_flash |
|------|--------|-----------|------------|-------------|

Build successes:
  * DISCO_F429ZI::GCC_ARM::MBED-BUILD

Build skips:
  * DISCO_F429ZI::GCC_ARM::TEST_APPS-DEVICE-EXAMPLEAPP
  * DISCO_F429ZI::GCC_ARM::TEST_APPS-DEVICE-SOCKET_APP
cleaning up Icetea log directory.
14:18:37.279 Using Icetea version 1.1.0
14:18:37.295 Loading suite from file
14:18:37.295 Importing testcases for filtering
14:18:37.358 Filtering testcases
14:18:37.358 Suite creation complete.
14:18:37.423 Test cases prepared.
14:18:37.424 Starting suite.
14:18:37.424 Starting repeat 1 of 1
14:18:37.424 Iteration 1 of 1
14:18:37.424 TC 'TCPSERVER_ACCEPT' will be skipped because of 'Test requiring application binary not build'
Cmd ('git rev-parse HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --abbrev-ref HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --show-toplevel') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git diff-index --quiet HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

14:18:37.700 Test case TCPSERVER_ACCEPT skipped.

14:18:37.700 Iteration 1 of 1
14:18:37.700 TC 'TCPSOCKET_ACCEPT' will be skipped because of 'Test requiring application binary not build'
Cmd ('git rev-parse HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --abbrev-ref HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --show-toplevel') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git diff-index --quiet HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

14:18:37.996 Test case TCPSOCKET_ACCEPT skipped.

14:18:37.996 Iteration 1 of 1
14:18:37.996 TC 'TCPSOCKET_BIND_PORT' will be skipped because of 'Test requiring application binary not build'
Cmd ('git rev-parse HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --abbrev-ref HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --show-toplevel') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git diff-index --quiet HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

14:18:38.278 Test case TCPSOCKET_BIND_PORT skipped.

14:18:38.278 Iteration 1 of 1
14:18:38.278 TC 'TCPSOCKET_ECHOTEST_BURST_SHORT' will be skipped because of 'Test requiring application binary not build'
Cmd ('git rev-parse HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --abbrev-ref HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --show-toplevel') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git diff-index --quiet HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

14:18:38.592 Test case TCPSOCKET_ECHOTEST_BURST_SHORT skipped.

14:18:38.592 Iteration 1 of 1
14:18:38.592 TC 'UDPSOCKET_BIND_PORT' will be skipped because of 'Test requiring application binary not build'
Cmd ('git rev-parse HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --abbrev-ref HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --show-toplevel') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git diff-index --quiet HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

14:18:38.888 Test case UDPSOCKET_BIND_PORT skipped.

14:18:38.888 Iteration 1 of 1
14:18:38.888 TC 'test_cmdline' will be skipped because of 'Test requiring application binary not build'
Cmd ('git rev-parse HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --abbrev-ref HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git rev-parse --show-toplevel') fails: fatal: Not a git repository (or any of the parent directories): .git

Cmd ('git diff-index --quiet HEAD') fails: fatal: Not a git repository (or any of the parent directories): .git

14:18:39.217 Test case test_cmdline skipped.

+--------------------------------+---------+-------------+---------------------------------------------+-----------+----------+
| Testcase                       | Verdict | Fail Reason |                 Skip Reason                 | platforms | duration |
+--------------------------------+---------+-------------+---------------------------------------------+-----------+----------+
| TCPSERVER_ACCEPT               |   skip  |             | Test requiring application binary not build |           |    0     |
| TCPSOCKET_ACCEPT               |   skip  |             | Test requiring application binary not build |           |    0     |
| TCPSOCKET_BIND_PORT            |   skip  |             | Test requiring application binary not build |           |    0     |
| TCPSOCKET_ECHOTEST_BURST_SHORT |   skip  |             | Test requiring application binary not build |           |    0     |
| UDPSOCKET_BIND_PORT            |   skip  |             | Test requiring application binary not build |           |    0     |
| test_cmdline                   |   skip  |             | Test requiring application binary not build |           |    0     |
+--------------------------------+---------+-------------+---------------------------------------------+-----------+----------+
+---------------+---------+
|    Summary    |         |
+---------------+---------+
| Final Verdict |   PASS  |
|     count     |    6    |
|    passrate   |  0.00 % |
|      skip     |    6    |
|    Duration   | 0:00:00 |
+---------------+---------+
14:18:39.342 Cleanup done.

C:\Users\ssir\Downloads\mbed-os-example-ble-master\BLE_Beacon>

-->
    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-01-04:
`[Error] cmd_ifconfig.cpp@27,2: #error [NOT_SUPPORTED] No network configuration found for this target.`

Missing target config for the test. You did not select which tests so all were built.

bridadan, 2019-03-04:
It appears that this test was compiled for a target that does not have the necessary prerequisites (network configuration). I'll close this issue, though please reopen this if you have further issues/questions.

closed 2019-03-04 by bridadan