ARMmbed/mbed-os #9178: Challenges with overriding linker scripts

unsignedint, 2018-12-21:
### Description

I am working on the NXP i.MX RT1050 EVK board which is set up for running code from hyper flash etc. We want to create a custom target that uses QSPI flash and a custom linker script, and goal is to make no changes to the mbed-os repository.

My feeling is that the best way to approach this is with a custom target that inherits from the `MIMXRT1050_EVK` target but replaces the linker script etc. That is all fine... but the challenge seems to be to select the correct linker script at link time.

(1)
It seems problematic to "remove" the existing default link script using the `extra_labels` options within target JSON definition, because of the current file hierarchy:

```
* targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MIMXRT1050
- device/... <- good stuff
- drivers/... <- more good stuff
- device/TOOLCHAIN_*/... <- linker scripts
- TARGET_EVK/... <- board-specific 
```

Because the linker script is under the `MIMXRT1050` root target directory we can't easily exclude it. *However*, if it was under the `EVK` directory where the board specific files are then we *can* easily exclude the `EVK` label in our config, and everything should be good.

(2)
Reading around it seems there is a hidden `-l` parameter accepted by `mbed compile` to explicitly pass a linker script. This is also reasonable, but it falls apart because the explicit linker script is passed as the _first_ linker file, and then in `toolchains/__init__.py` it selects the _last_ linker file (L625):

```python
        try:
            linker_script = [path for _, path in r.get_file_refs(FileType.LD_SCRIPT)
                             if path.endswith(self.LINKER_EXT)][-1]
        except IndexError:
            raise NotSupportedException("No linker script found")
```

Another way to fix this is to make the `-l` option override this.

I am created this issue to seek some feedback from the mbed team about the best way to solve this problem, and are happy to make the necessary change if we have a direction.

### Issue request type

    [X] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2018-12-21:
There's similar issue already reported:  https://github.com/ARMmbed/mbed-os/issues/7126, will this help?

maclobdell, 2019-09-09:
Can this issue be closed?  It is very old.  Was this resolved?

0xc0170, 2019-09-10:
I don't think so. Only workaround to copy the structure and overwrite it. 

> Because the linker script is under the MIMXRT1050 root target directory we can't easily exclude it. However, if it was under the EVK directory where the board specific files are then we can easily exclude the EVK label in our config, and everything should be good.

This could be fixed if its still causing an issue - move linker script to be more specific for a board.