ARMmbed/mbed-os #9189: Incorrect handling of build system components

screamerbg, 2018-12-23:
### Description

The build system incorrectly handles a use case where `"component_add": ["SD"]` is "emitted" from a library, e.g. `FEATURE_BOOTLOADER/mbed_lib.json`. The build system will correctly add `-DCOMPONENT_SD=1` macro, but it will not add relevant component folders to the include paths, thus fails to compile the codebase. 

To reproduce:
1. `mbed import http://github.com/screamerbg/pelion-integration`
2. `mbed update 48bd908` - don't use master which already has a workaround in place for this issue.
3. `mbed compile -t ARM -m UBLOX_C030_U201 -c -v`
4. Observe that `-DCOMPONENT_SD=1` is in the reported macros on the console, but the `mbed-os/components/COMPONENT_SD` path doesn't exist in the includes file in `BUILD\UBLOX_C030_U201\ARM\.includes_<hash>.txt`
5. Edit `mbed_app.json` and add `"target.components_add"             : ["SD"],` for entry `UBLOX_C030_U201`
6. `mbed compile -t ARM -m UBLOX_C030_U201 -c -v`
7. Observe that once again `-DCOMPONENT_SD=1` is in the reported macros on the console, but this time the `mbed-os/components/COMPONENT_SD` path correctly is added to the includes file in `BUILD\UBLOX_C030_U201\ARM\.includes_<hash>.txt`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

screamerbg, 2019-01-29:
@cmonr Correct, although you can reproduce it with any code where `target.components_add` is "emitted" from a library (mbed_lib.json), but the application config (mbed_app.json) doesn't declare it, resulting the behavior explained above.

I've discussed this with @theotherjimmy in person about couple of weeks ago.

screamerbg, 2019-03-28:
Bump @theotherjimmy @cmonr

adbridge, 2020-10-14:
As this applies to the old tools I'm going to close this. @Patater can you add this to your list of things to ensure it isn't an issue with the new tools...

closed 2020-10-14 by adbridge