ARMmbed/mbed-os #9194: [Seeed Wio 3G] cellular function is not working in Mbed OS 5.11

toyowata, 2018-12-25:
### Description

The Wio 3G has the Quectel UG96 module on the board and cellular function was working up to the Mbed OS 5.10.4, but it does not seem to work in Mbed OS 5.11.

#### Result of mbed-os-example-cellular

[Mbed OS 5.10.x]
Version
```
mbed-os-example-cellular (#b7e50ec6249e, tag: mbed-os-5.10.3)
`- mbed-os (#bf6f2c3c6434, tags: mbed-os-5.10.3, mbed_lib_rev164)
```

Test result
```
-example-cellular
Establishing connection
..........

Connection Established.
TCP: connected with echo.mbedcloudtesting.com server
TCP: Sent 4 Bytes to echo.mbedcloudtesting.com
Received from echo server 4 Bytes

Success. Exiting
```
[Mbed OS 5.11.0]
Version
```
mbed-os-example-cellular (#aeae61fdb1d6, tag: mbed-os-5.11.0-rc2)
`- mbed-os (#d9625811c919, tag: mbed-os-5.11.0-rc2)
```

Test result
```
-example-cellular
Establishing connection
..............................
```

More detail logs with trace
```
0ms][INFO][CELL]: New CellularContext  (0x20003e98)
[00000000ms][INFO][CELL]: CellularContext plmn (null)
[00000003ms][INFO][CELL]: CellularContext connect
[00000008ms][INFO][CELL]: Start connecting (timeout 1 s)
AT
[00001015ms][WARN][CELL]: AT timeout
AT
[00001015ms][WARN][CELL]: AT timeout
[00001015ms][INFO][CELL]: Modem power ON (timeout 1 s)
ATE0
[00002816ms][WARN][CELL]: AT timeout
[00002816ms][WARN][CELL]: AT timeout
[00002816ms][INFO][CELL]: Continue after 2 seconds
ATE0
[00005817ms][WARN][CELL]: AT timeout
[00005817ms][WARN][CELL]: AT timeout
[00005817ms][INFO][CELL]: Continue after 4 seconds
ATE0
[00010818ms][WARN][CELL]: AT timeout
[00010818ms][WARN][CELL]: AT timeout
[00010818ms][INFO][CELL]: Continue after 8 seconds
ATE0
[00019819ms][WARN][CELL]: AT timeout
[00019819ms][WARN][CELL]: AT timeout
[00019819ms][INFO][CELL]: Continue after 16 seconds
ATE0
[00036820ms][WARN][CELL]: AT timeout
[00036820ms][WARN][CELL]: AT timeout
[00036820ms][INFO][CELL]: Continue after 32 seconds
ATE0
[00069821ms][WARN][CELL]: AT timeout
[00069821ms][WARN][CELL]: AT timeout
[00069821ms][INFO][CELL]: Continue after 64 seconds
ATE0
[00134822ms][WARN][CELL]: AT timeout
[00134822ms][WARN][CELL]: AT timeout
[00134822ms][INFO][CELL]: Continue after 128 seconds
ATE0
[00263823ms][WARN][CELL]: AT timeout
[00263823ms][WARN][CELL]: AT timeout
[00263823ms][INFO][CELL]: Continue after 600 seconds
ATE0
[00864824ms][WARN][CELL]: AT timeout
[00864824ms][WARN][CELL]: AT timeout
[00864824ms][INFO][CELL]: Continue after 1200 seconds
[01800008ms][WARN][CELL]: No cellular connection

Couldn't connect: -3019, will retry
[01800008ms][INFO][CELL]: CellularContext connect
ATE0
[02065825ms][WARN][CELL]: AT timeout
[02065825ms][WARN][CELL]: AT timeout
[02065825ms][ERR ][CELL]: CellularStateMachine failure: Device ready
[02065825ms][ERR ][CELL]: CellularStateMachine target state Attaching network, current state Device ready
AT+CGDCONT?
[02066826ms][WARN][CELL]: AT timeout
[02066826ms][WARN][CELL]: AT timeout
[02066826ms][ERR ][CELL]: Failed to activate network context! (-3004)

Couldn't connect: -3004, will retry
[02066826ms][INFO][CELL]: CellularContext connect
```

#### Result of automated tests for cellular

[Mbed OS 5.10.4]
Version
```
mbed-os-5.10.4 (#2fd0c5cfbd83, tags: latest, mbed-os-5.10.4)
```
[mbed-os-5.10.4_wio_3g_test.txt](https://github.com/ARMmbed/mbed-os/files/2708113/mbed-os-5.10.4_wio_3g_test.txt)

[Mbed OS 5.11.0]
Version
```
mbed-os-5.11.0 (#6a0a86538c0b, tags: latest, mbed-os-5.11.0)
```
[mbed-os-5.11.0_wio_3g_test.txt](https://github.com/ARMmbed/mbed-os/files/2708112/mbed-os-5.11.0_wio_3g_test.txt)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cc @ARMmbed/team-seeed @MarceloSalazar

cmonr, 2018-12-25:
@ARMmbed/mbed-os-wan

toyowata, 2019-01-17:
The cause of this problem is HW issue (CTS/RTS connection is wrongly swapped between MCU and UG96 module).  Temporary workaround is to disable flow control in the `mbed_app.json`.

```json
        "WIO_3G": {
            "target.device_has_remove": ["SERIAL_FC"]
        }
```

closed 2019-01-17 by toyowata