ARMmbed/mbed-os #9201: BLE adding multiple services data to advertising packet

desmond-blue, 2018-12-27:
### Description

For building BLE advertising packet, the API AdvertisingDataBuilder::addData() would be used to add AD elements. This API only allows each AD type being added once, however there are requirements for adding each AD type more than once, like multiple services data.

Currently we can use [this API](https://github.com/ARMmbed/mbed-os/blob/995725b50a7538e3f0b331823c0bd40a5691d9a5/features/FEATURE_BLE/ble/gap/Gap.h#L590-L593) to set any payload, and it would be easier for user to organize advertising data packets if we update the data builder.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

cmonr, 2019-01-03:
@ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2019-01-03:
addData is supported by:
    AdvertisingDataBuilder::replaceData
    AdvertisingDataBuilder::appendData
    AdvertisingDataBuilder::removeData
    AdvertisingDataBuilder::addOrReplaceData
    AdvertisingDataBuilder::addOrAppendData