ARMmbed/mbed-os #9213: STM32F4 Serial Fail

masoudr, 2018-12-29:
### Description
I'm using DISCO F4 board (STM32f407VG) and a Quectel GSM modem to create a two-way serial communication to send AT commands from PC to vice-versa. I'm using default mbed Serial API but I only get 2 or 3 chars of any answer from the modem. I tried some buffered libraries like `MODSERIAL` and they worked fine.
Here is an example:
```
#include "mbed.h"

Serial pc(PA_9, PA_10);
Serial modem(PA_2, PA_3);

using namespace std;

string readSerial()
{
    string output = "";
    while (modem.readable())
    {
        output += modem.getc();
    }
    return output;
}

bool Modem::writeSerial(const char *input)
{
    if (modem.writeable())
    {
        modem.printf("%s\r\n", input);
        return true;
    }
    return false;
}

void direct_command(){
    string temp;
    if (pc.readable()){
        pc.gets(temp, sizeof(temp));
        strcat(temp, "\r\n");
        writeSerial(temp);
    }
    temp = readSerial();
    if(!temp.empty())
        printf("ANSWER: %s\r\n", temp.c_str());
}

int main()
{
    pc.baud(115200);
    modem.baud(115200);
    
    printf("Setup Complete!\n");
    while (1)
    {
        direct_command();
    }
}

```
I get something like this in console output:

> ANSWER:
> ANSWER: E
> ANSWER:
> ANSWER: d
> ANSWER:
> ANSWER: y

### Issue request type

    [ ] Question
    [ ] Enhancement
    [*] Bug

0xc0170, 2019-01-02:
cc @ARMmbed/team-st-mcd

kjbracey-arm, 2019-01-02:
An instantaneous `readable()` check is never going to be useable as a "should I finish reading" test. You'll need to have a proper framing mechanism, or if you don't have that, something with a finite timeout. You need to be waiting at least long enough for the next character to arrive.

`Serial` doesn't really handle input well - use `UARTSerial`, or if actually talking to the console, just use `stdin` and `stdout`.

I would write your example as something like this, assuming that `pc` is your console/stdin/stdout.

```
#include "mbed.h"

static UARTSerial modem(PA_2, PA_3);
static FILE *modem_stream;

using namespace std;

void direct_command(){
    char buffer[256];
    gets(buffer);
    fprintf(modem_stream, "%s\r", buffer); // `\r` is usually what modems want on input
    fgets(buffer, sizeof buffer, modem_stream);
    printf("ANSWER: %s\n", buffer);
}

int main()
{
    modem.set_baud(115200);
    modem_stream = fdopen(&modem, "w+");
    
    printf("Setup Complete!\n");
    while (1)
    {
        direct_command();
    }
}
```

Eliminating the readable/writable checks will make it reliable, at least as long as you always get a 1-line response per 1 line sent. Doing it properly, with timeouts and recovery, needs a lot more work - I'd suggest looking at `ATCmdParser` to handle the modem for you.

kjbracey-arm, 2019-01-02:
Alternatively, if you just want a "raw data pump" between two serial ports, rather than something specifically line-based, there is an example application in https://github.com/ARMmbed/mbed-os-5-docs/pull/745/files ("Poll example: Transfer bidirectional data between two ports, acting as a virtual link.")

masoudr, 2019-01-04:
@kjbracey-arm Thanks the first option worked for me.

closed 2019-01-04 by masoudr