ARMmbed/mbed-os #9216: Defining MBED_SLEEP_TRACING_ENABLED caused error

syhoangmai, 2018-12-31:
### Description

I am optimizing power consumption for my device and following this api reference https://os.mbed.com/docs/v5.11/apis/power-management.html. I define the macro MBED_SLEEP_TRACING_ENABLED to get tracing info about locks preventing sleep mode, however doing so will broke the program entirely. 

On NUCLEO-F767ZI and  DISCO-F746NG, both stuck at Ethernet initialization. mbed-os is at release 5.11.0 (6a0a865). Compiler: online compiler and gcc arm 6 2017-q2-update. It got stuck at this message, either reset or just stuck here 
```
Sleep locks held:
[id: stm32xx_emac.cpp, count: 1]
[id: mbed_rtx_idle.cpp, count: 1]
UNLOCK: mbed_rtx_idle.cpp, ln: 137, lock count: 1
LOCK: mbed_rtx_idle.cpp, ln: 135, lock count: 2
```

Here is a simple program that should be able to reproduce this issue https://os.mbed.com/users/sh114203/code/Ethernet-try-power-tracing/.

I also tried this with UBLOX_C030_N211. At cellular modem initialization the program just halt and give this error. So the problem might not comes from the Ethernet.
```
Location: 0x800C611
Error Value: 0x2
Current Thread: main  Id: 0x20005494 Entry: 0x800D4D7 StackSize: 0x1388 StackMem: 0x200054D8 SP: 0x20006210
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x8001012F
-- MbedOS Error Info --
```
All these problems gone if MBED_SLEEP_TRACING_ENABLED is not defined.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-01-02:
The error means "Operation Prohibited in ISR context" . Can you find the code that is triggering this?

cc @ARMmbed/mbed-os-ipcore

0xc0170, 2019-01-02:
> It seems it should instead use mbed_error_printf to get interrupt-safe printing.

Will move this ticket to the core team.

deepikabhavnani, 2019-01-04:
> Seems like `sleep_manager_sleep_auto` (called with ISR disabled) is calling `debug`, which calls `vfprintf` (claims a mutex).
> 
> It seems it should instead use `mbed_error_printf` to get interrupt-safe printing.

Yes all the `debug` calls in sleep manager for sleep tracing should be changed to `mbed_error_printf` since most sleep API's and calls are interrupt and thread safe.

Issue was seen only with `debug` profile and was not captured in `develop` profile because `MBED_TRAP_ERRORS_ENABLED` flag was not set in `develop` profile.

Fix in https://github.com/ARMmbed/mbed-os/pull/9260

closed 2019-01-10 by cmonr