ARMmbed/mbed-os #9228: STM32 st-link reflash does not work in Mac OS X

jussisip, 2019-01-03:
### Description

Any Mbedos Greentea test fails after reflashing the board. Flashing and single test will run successfully only once after power on / off for the board. Repeating the same test case run without power on / off is not possible.

Issue prevents to run Greentea tests in a row for Mbed OS locally with Mac OS X. Power cycling on / off is required between test cases in order to run them successfully.

Workaround for at least Linux seems to exist which is shortly described by st-link firmware readme.txt.

Test configuration
* Macbook Pro, MacOS Mojave
* Boards: DISCO_L475VG_IOT01
* ST-link FW: STSW-LINK007 2.32.22
* Greentea 1.4.0
* GCC Arm
* Mbedos sha: 51143eab0

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

jeromecoutant, 2019-01-03:
Hi

We are currently working on some ST Link FW issues.
We expect some patches soon.

@schstm

jussisip, 2019-01-23:
@jeromecoutant Do you any updates available yet for this issue?

jeromecoutant, 2019-01-28:
Hi all

The new
https://www.st.com/en/development-tools/stsw-link007.html

studavekar, 2019-01-30:
@jeromecoutant is there a release doc providing details of features added/ bug fixed?  was not able to find one.

jeromecoutant, 2019-02-05:
@jussisip Should we close the issue ?

jeromecoutant, 2019-03-06:
@ARMmbed/mbed-os-maintainers  could you close it ?
Thx

closed 2019-03-06 by cmonr