ARMmbed/mbed-os #9235: K64F, K82F: failing storage tests when SD component is not available

mprse, 2019-01-03:
### Description

Issue found on master:
51b8d6e59d (HEAD -> master, upstream/master) Merge pull request #9187 from deepikabhavnani/iar8_build_failures

If the `SD` component is removed from components list of `K64F` and `K82F` targets in `targets.json` configuration file, then the following tests fail:

Test | Board | Compiler | Result
-- | -- | -- | --
features-storage-tests-blockdevice-general_block_device | K82F, K64F | ARM, GCC_ARM, IAR | failed
features-device_key-tests-device_key-functionality | K82F, K64F | ARM, GCC_ARM, IAR | failed
features-storage-tests-kvstore-static_tests | K82F, K64F | ARM, GCC_ARM, IAR | failed
features-storage-tests-kvstore-general_tests_phase_1 | K82F, K64F | ARM, GCC_ARM, IAR | failed
features-storage-tests-kvstore-securestore_whitebox | K82F, K64F | ARM, GCC_ARM, IAR | failed
tests-psa-prot_internal_storage | K64F | ARM, GCC_ARM, IAR | failed

These tests store the data on the storage device. If `SD` component is available, then SD card is used as a storage device. If `SD` component is not available then Flash is used instead since `FLASHIAP` component is available also for both boards. 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-01-03:
cc @ARMmbed/team-nxp

mmahadevan108, 2019-01-11:
While investigating I came across a potential issue with the test. The below line is trying to program without doing an erase. This could potentially fail on flash locations that are not in the erase state. 
https://github.com/ARMmbed/mbed-os/blob/master/features/storage/TESTS/blockdevice/general_block_device/main.cpp#L366

Kindly review the other test failures to see if this could be the cause for failures seen with Flash operation

0xc0170, 2019-01-14:
> Kindly review the other test failures to see if this could be the cause for failures seen with Flash operation

@ARMmbed/mbed-os-storage Please review

offirko, 2019-01-14:
As @mmahadevan108 analyzed, the "general_block_device: test is failing due to test bug of programing prior to erasing.  It will be fixed - Thanks !

The rest of the tests are no longer failing on updated master. They failed because the storage type did not match the selected component (FlashIAP). A recent change in mbed-os made the storage type selected **according to the component defined in targets.json** so it no longer fails.

PR #9377 (2019-01-18): test: general_block_device fix to always erase prior the first program
### Description

general_block_device test fixed to always erase prior to first program.

Fix: K64F, K82F: failing storage tests when SD component is not available
Resolves #9235
Internal Jira reference: https://jira.arm.com/browse/MBOCUSTRIA-380

### Pull request type
    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
closed 2019-01-18 by 0xc0170