ARMmbed/mbed-os #9240: UBLOX_EVK_ODIN_W2 fails to initialize RTC in TLSSocket example

michalpasztamobica, 2019-01-03:
### Description

Target: UBLOX_EVK_ODIN_W2
Toolchain: GCC_ARM, ARM, IAR
Tools: Just mbed-cli and this example: https://github.com/ARMmbed/mbed-os-example-tls-socket/
SHA of mbed-os: [51b8d6e](https://github.com/ARMmbed/mbed-os/commit/51b8d6e59d17748ca477505e77a44a848ceef4c4)

Steps to reproduce:
1) mbed import https://github.com/ARMmbed/mbed-os-example-tls-socket.git
2) Add "target.network-default-interface-type": "ETHERNET" to the mbed_app.json
3) mbed compile -m UBLOX_EVK_ODIN_W2 -t GCC_ARM --flash
4) Run serial terminal (at 9600 bps by default)
5) See the following error:
```
TLSSocket Example.                                                        
Mbed OS version: 5.11.1                                                   
                                                                          
Connecting to network
[DBG ][TLSW]: mbedtls_ssl_conf_ca_chain()
Connecting to api.ipify.org
[INFO][TLSW]: Starting TLS handshake with api.ipify.org                         
[DBG ][TLSW]: mbedtls_ssl_setup()                                               
                                                                                
                                                                                
++ MbedOS Error Info ++                                                         
Error Status: 0x80FF0100 Code: 256 Module: 255                                  
Error Message: Fatal Run-time error                                             
Location: 0x80309FD                                                             
Error Value: 0x0                                                                
Current Thread: main  Id: 0x20010694 Entry: 0x802B25B StackSize: 0x2000 StackMe 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0100  
-- MbedOS Error Info --                                                         
Cannot initialize RTC with LSE 
```

In other scenarios I could also see different failures within RTC initialization or SettingDate/Time in RTC.

The issue is gone if I comment the [MBEDTLS_HAVE_TIME definition](https://github.com/ARMmbed/mbed-os/blob/51b8d6e59d17748ca477505e77a44a848ceef4c4/features/mbedtls/inc/mbedtls/config.h#L151) in [mbedtls/config.h](https://github.com/ARMmbed/mbed-os/blob/master/features/mbedtls/inc/mbedtls/config.h)

Other platforms (K64F, K66F, NUCLEO_F429ZI) don't behave like this so I would assume the issue is related to UBLOX's RTC.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

michalpasztamobica, 2019-01-03:
In case it helps, here is a backtrace of how the code gets from TLSSocket through mbedtls down to the RTC initialization:
```
#0  rtc_init () at ./mbed-os/targets/TARGET_STM/rtc_api.c:49
#1  0x0802e0f0 in set_time () at ./mbed-os/platform/mbed_rtc_time.cpp:111
#2  0x0802e138 in time () at ./mbed-os/platform/mbed_rtc_time.cpp:91
#3  0x0801c574 in ssl_generate_random () at ./mbed-os/features/mbedtls/src/ssl_cli.c:735
#4  ssl_write_client_hello () at ./mbed-os/features/mbedtls/src/ssl_cli.c:861
#5  0x0801cfc2 in mbedtls_ssl_handshake_client_step ()
    at ./mbed-os/features/mbedtls/src/ssl_cli.c:3612
#6  0x080220de in mbedtls_ssl_handshake_step () at ./mbed-os/features/mbedtls/src/ssl_tls.c:8309
#7  0x0802212e in mbedtls_ssl_handshake () at ./mbed-os/features/mbedtls/src/ssl_tls.c:8333
#8  0x0802a4c2 in TLSSocketWrapper::continue_handshake ()
    at ./mbed-os/features/netsocket/TLSSocketWrapper.cpp:227
#9  0x0802abdc in TLSSocketWrapper::start_handshake ()
    at ./mbed-os/features/netsocket/TLSSocketWrapper.cpp:202
```

jamesbeyond, 2019-01-09:
This is a know issue for Ulbox. basically the RTC require a second oscillator which is not connected by default. You need short the Jumper21 on Odin board to get it connected, that will get ride of the crash.
for more info, you can check ticket IOTPART-6580 on Jira for more info. @michalpasztamobica

michalpasztamobica, 2019-01-10:
I confirm - when I inserted the jumper the error is gone. Thank you, @jamesbeyond !

Out of curiosity... my UBLOX_EVK_ODIN_W2 board has an STM32F103CBT6 microcontroller. According to the [datasheet](https://www.st.com/resource/en/datasheet/cd00161566.pdf) it has a "32 kHz oscillator for RTC". In the RTC chapter 2.3.14 of this document I also read that both an internal or external oscillator can be used to run RTC. Could you help me understand why an external oscillator is required?

jamesbeyond, 2019-01-10:
It is a Low Power Oscillator. it was designed to be used when wifi chip go into lowpower mode. based on on the [UserGuide](https://www.u-blox.com/sites/default/files/EVK-ODIN-W2_UserGuide_%28UBX-16007132%29.pdf). but they also mentioned if the LPO is not connected, it will try to use the MCU internal one. So It seems Ublox design changed STM MCU RTC behaviour. As for the reason, I am not very clear. maybe @ARMmbed/team-ublox have some explanations

aqib-ublox, 2019-01-15:
@jamesadevine thanks for you concern but let me brief you little bit. Actually RTC should use internal oscillator if external not available but for that we have to program RTC hardware configuration register to use clk source from LSI rather than LSE.

In mbed-os we are configuring RTC to use LSE by default please see target.json "lse_available": 1 by default

>    "FAMILY_STM32": {
        "inherits": ["Target"],
        "public": false,
        "extra_labels": ["STM"],
        "supported_toolchains": ["ARM", "uARM", "IAR", "GCC_ARM"],
        "macros": ["TRANSACTION_QUEUE_SIZE_SPI=2"],
        "config": {
            "lse_available": {
                "help": "Define if a Low Speed External xtal (LSE) is available on the board (0 = No, 1 = Yes). If Yes, the LSE will be used to clock the RTC, LPUART, ... otherwise the Low Speed Internal clock (LSI) will be used",
                "value": "1"
            },

(ODIN inherits this config) which means during initialization target conifg instructing our code to program RTC to use LSE. If LSE is not available we need to override macro like 

        "overrides": {"lse_available": 0},
under UBLOX_EVK_ODIN_W2 

>     "UBLOX_EVK_ODIN_W2": {
        "inherits": ["MODULE_UBLOX_ODIN_W2"],
        "supported_form_factors": ["ARDUINO"],
        "release_versions": ["5"],
        "device_has_remove": [],
        "overrides": {"lse_available": 0},

and recompile code. this would enable code to progam RTC to use LSI.