ARMmbed/mbed-os #9242: NRF52 BLE idle current consumption higher than expected

melvinvdb, 2019-01-03:
### Description

Target: NRF52
Toolchain: GCC_ARM
Tools: Mbed OS 5.11 mbed-cli 1.8.3
mbed-os-example-ble: BLE_Beacon

#### Issue:
When using BLE, the current consumption is not as low as expected (14uA instead of ~3uA). The current is higher than the Nordic SDK example.

#### Steps to reproduce:
Used a SparkFun nRF52832 Breakout so only a NRF52832 is connected (this is also happening on my custom NRF52 board).
The LF clock is configured as RC so no external 32khz is used! If I remember correctly this issue also happens when using an external 32khz clock.
Did some current measurements with an Otii Arch.

When using nRF5_SDK_11.0.0_89a8197 ble_app_beacon example:
[image]
Zoomed in:
[image]
As you can see the idle current is around 2.38uA.

Now when using the BLE_Beacon example:
I've disabled `printMacAddress()` so no serial interface is initialized. 
[image]
A average idle current of around 14uA is seen.

As you can see the current peaks remain the same except for Mbed which is happening way faster. This result in a pretty large increase in current consumption.

Anyone knows why this is happening and how to solve this issue? 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

desmond-blue, 2019-01-25:
It's related to #9093.

desmond-blue, 2019-07-30:
Hi @melvinvdb , we believe this issue has been resolved, would you confirm and close this issue? Thanks.

closed 2019-08-14 by 0xc0170