ARMmbed/mbed-os #9255: FLASHIAP component tests failed

jeromecoutant, 2019-01-04:
### Description

With #8667 FLASHIAP component has been added for DISCO_L475VG_IOT01A and DISCO_L476VG

Tests are FAIL with 5.11.0

| target                      | platform_name       | test suite                                                                   | test case                                                                                         | passed | failed | result  | elapsed_time (sec) |
|-----------------------------|---------------------|------------------------------------------------------------------------------|---------------------------------------------------------------------------------------------------|--------|--------|---------|--------------------|
| DISCO_L475VG_IOT01A-ARM     | DISCO_L475VG_IOT01A | components-storage-blockdevice-component_flashiap-tests-filesystem-fopen     | FSLITTLE_FOPEN_TEST_00: format() test.                                                            | 0      | 1      | FAIL    | 0.11               |
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | components-storage-blockdevice-component_flashiap-tests-filesystem-fopen     | FSLITTLE_FOPEN_TEST_00: format() test.                                                            | 0      | 1      | FAIL    | 0.11               |
| DISCO_L475VG_IOT01A-IAR     | DISCO_L475VG_IOT01A | components-storage-blockdevice-component_flashiap-tests-filesystem-fopen     | FSLITTLE_FOPEN_TEST_00: format() test.                                                            | 0      | 1      | FAIL    | 0.11               |
| DISCO_L476VG-ARM     | DISCO_L476VG  | components-storage-blockdevice-component_flashiap-tests-filesystem-fopen     | FSLITTLE_FOPEN_TEST_00: format() test.                                                            | 0      | 1      | FAIL    | 0.11               |
| DISCO_L476VG-GCC_ARM | DISCO_L476VG  | components-storage-blockdevice-component_flashiap-tests-filesystem-fopen     | FSLITTLE_FOPEN_TEST_00: format() test.                                                            | 0      | 1      | FAIL    | 0.11               |
| DISCO_L476VG-IAR     | DISCO_L476VG  | components-storage-blockdevice-component_flashiap-tests-filesystem-fopen     | FSLITTLE_FOPEN_TEST_00: format() test.                                                            | 0      | 1      | FAIL    | 0.11               |

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-01-04:
cc @ARMmbed/mbed-os-storage

jeromecoutant, 2019-01-09:
Fixed by #9268

closed 2019-01-09 by jeromecoutant