ARMmbed/mbed-os #9263: BLE: Need API to get Data Length Extension status if it's Bluetooth 4.2 or above.

kaiser-ren, 2019-01-06:
### Description
Data Length Extension (DLE) was introduced in Bluetooth v4.2, but this feature is optional which means that some kits are 4.2, but they don't have DLE feature. 
So developer need API to get DLE status, when has DLE status, it's possible to estimate the top-proof of the connection throughput. 

### Issue request type

    [ ] Question
    [x ] Enhancement
    [ ] Bug

0xc0170, 2019-01-07:
cc @ARMmbed/mbed-os-pan

pan-, 2019-01-07:
@kaiser-ren We plan to introduce something a bit different: right after connection we want to initiate an MTU negotiation between the ends of the connection. The GATT client (I'm not so sure for server) APIs already take advantage of the MTU size negotiated so there is nothing else to worry about from a developer perspective. 

@paul-szczepanek-arm Would it be possible to expose a public API to get the MTU size of a given connection ? Or we can fire an event when the MTU size change.

kaiser-ren, 2019-01-08:
@pan- Thanks for the update. It would be awesome to have APIs to get negotiated ATT_MTU size, but it's not enough for developers to calculated throughput. For example when ATT_MTU is 185 bytes:
* if each ATT packet length is 27 bytes, for 185 bytes MTU, it needs **7** packets; 
* if each ATT packet length is 247 bytes (like I mentioned before, DLE is enabled), for 185 bytes MTU, it just needs **one** packet. 
So, data packet length for each ATT packet has huge impact for BLE connection throughput, meanwhile, this is also the most common question about BLE: how fact can my peripheral be with a smartphone?

kaiser-ren, 2019-01-08:
And I just did a test between Nordic nRF52840 and iPhone8:
*connection interval is 50ms
*ATT_MTU is 185 byte
The difference of DLE-enable and DLE-disable:
* when I set packet length is 27 bytes (DLE diable), the throughput is about 48.27 kbps; 
* when I set packet length is 247 bytes (DLE enable), the throughput is about 135 kbps; 

So, API about data packet length is important.

pan-, 2019-01-08:
@kaiser-ren We intend to provide both information: packet size and ATT_MTU size in the form of events as the value is negotiated by both ends. DLE negotiation is enabled by default on ports using the cordio stack; it should be enabled for builds using the Cordio link layer on Nordic targets. For softdevice build I don't think it is enabled yet.

kaiser-ren, 2019-01-19:
Hello @pan- ,
As for this issue, any update on it? I'd like to check with you that when it could be on the schedule. 
Thanks.

pan-, 2019-01-30:
@kaiser-ren Please have a look at #9537 .

paul-szczepanek-arm, 2019-02-15:
Hi @kaiser-ren, the MTU PR has been merged to master. As for your original question you can check features like Data Length Extension with Gap::isFeatureSupported()

kaiser-ren, 2019-02-15:
Hi @paul-szczepanek-arm , thank you so much to notify me about this, well done!
I will test it later in next week!

closed 2019-02-15 by kaiser-ren