ARMmbed/mbed-os #9265: mbed-os-example-ble-BatteryLevel, can be compiled on DISCO-L475VG-IOT01A, failed on NRF52_DK

kaiser-ren, 2019-01-06:
### Description

I use *mbed-cli* to compile **mbed-os-example-ble-BatteryLevel ** sample code, mbed-os is **v5.10.4**, compiler is **GCC_ARM**
I can compile the sample code when board is **DISCO-L475VG-IOT01A**. 
But failed when target is **NRF52_DK**, the error is:

> Link: mbed-os-example-ble-BatteryLevel_application
Elf2Bin: mbed-os-example-ble-BatteryLevel_application
b'C:\\Program Files (x86)\\GNU Tools ARM Embedded\\8 2018-q4-major\\bin\\arm-none-eabi-objcopy: .\\BUILD\\NRF52_DK\\GCC_ARM\\mbed-os-example-ble-BatteryLevel_application.hex 64-bit address 0x4b4fa300023000 out of range for Intel Hex file'
[ERROR] can't concat str to bytes
[mbed] ERROR: "c:\python37\python.exe" returned error.
       Code: 1
       Path: "C:\Users\xxx\work\Github\Mbed\mbed-os-example-ble-BatteryLevel"
       Command: "c:\python37\python.exe -u C:\Users\xxx\work\Github\Mbed\mbed-os-example-ble-BatteryLevel\mbed-os\tools\make.py -t GCC_ARM -m NRF52_DK --source . --build .\BUILD\NRF52_DK\GCC_ARM"

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

0xc0170, 2019-01-07:
cc @ARMmbed/mbed-os-pan

kaiser-ren, 2019-01-08:
Just verified, I used incorrect compiler, the target GCC should be v6 q2 2017.

closed 2019-01-08 by kaiser-ren