ARMmbed/mbed-os #9270: Cellular usage (and information) on mbed-os 5.11.x

vidavidorra, 2019-01-07:
### Description

In mbed-os v5.11.0 the `EasyCellularConnection` class has disappeared without direct mention of that in the release logs, other than major cellular refactoring, so I'm guessing it was part of that. Anyhow, I'm now using the `CellularBase` instead, but tried `CellularContext` too, which somehow takes a lot longer to create a cellular connection. I'm using it with the mbed-os PPP and an u-blox SARA-G350 and `#define CELLULAR_DEVICE UBLOX_PPP`

With the `EasyCellularConnection` I was able to read the u-blox device and SIM information, by getting the `CellularDevice` using `::get_device()` and file handle for `CellularInformation` using `::get_serial()` and `::get_sim()` for the SIM information. This was using the snippet below, where `_connection` is an instance pointer of `EasyCellularConnection`.
- u-blox module SN (IMEI)
- SIM ICCID
- SIM IMSI

Now, the problem is that with the `CellularBase`, and `CellularContext` for that matter, I'm not able to use this code anymore. I can use `CellularDevice::get_default_instance()` to get the `CellularDevice`, which seems to be a valid pointer but there is no function to get a file handle for passing to `open_information`. I've added the small snippet below to the `CellularDevice` class, which seems to return a valid pointer, but the `get_serial_number` function just hangs for 30 sec and then returns `-3012`, same goes for the SIM functions. Passing no file handle, so it uses the default, has the same result.

```c++
    CellularDevice *device = _connection->get_device();
    CellularInformation *information = device->open_information(_connection->get_serial());
    FileHandle *fh = device->get_file_handle();
    TRACE_DEBUG("Got file handle: (%p) %p", device, fh);
    CellularInformation *information = device->open_information(fh);

    if (information == nullptr) {
        TRACE_DEBUG("No information!!!");
    } else {
        char model[9 + 1];
        nsapi_error_t ret = information->get_serial_number(model, sizeof(model),
                                                           mbed::CellularInformation::SerialNumberType::SN);
        device->close_information();
        if (ret != NSAPI_ERROR_OK) {
            TRACE_DEBUG("Cellular model not found (%d)!", ret);
        } else {
            TRACE_DEBUG("Cellular model is: '%s'", model);
        }
    }
    device->close_information();

    CellularSIM *sim = _connection->get_sim();
    if (sim == nullptr) {
        TRACE_DEBUG("No SIM!!!");
    } else {
        char iccid[100];
        char imsi[100];
        nsapi_error_t imsi_ret = sim->get_imsi(imsi);
        TRACE_DEBUG("SIM IMSI (%d): '%s'", imsi_ret, imsi);
        nsapi_error_t iccid_ret = sim->get_iccid(iccid, 99);
        TRACE_DEBUG("SIM ICCID (%d): '%s'", iccid_ret, iccid);
    }
    device->close_sim();
```

Now, two questions.
1. What am I suppose to use for cellular connection: `CellularBase`, `CellularContext` or something else?
    - Note the difference in behaviour that `CellularContext` takes significantly longer to make a connection (over 2 minutes) than `CellularBase` (less tan 30 seconds).
    - The [mbed-os-example-cellular](https://github.com/ARMmbed/mbed-os-example-cellular) uses `CellularBase`.
2. How am I supposed to get the u-blox module and SIM information. What I'm interested in is the IMEI, SIM IMSI and SIM ICCID.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-01-07:
> In mbed-os v5.11.0 the EasyCellularConnection class has disappeared without direct mention of that in the release logs, other than major cellular refactoring, so I'm guessing it was part of that.

Is this refactoring well explained in the git history? What did help you to refactor your code? I am interested in docs improvements.

cc @ARMmbed/mbed-os-wan

vidavidorra, 2019-01-07:
@0xc0170 Well, I just looked at the source code in `features/cellular/framework/API`. That's what kind of helped me, later I saw something about the cellular example in one of the Git tickets so took a look at that and saw it uses `CellularBase`. I haven't looked at the git history too much, more at the release notes, but that doesn't directly mention anything about the `EasyCellularConnection`. Even more so, the docs in the source folder [features/cellular](https://github.com/ARMmbed/mbed-os/tree/master/features/cellular) are super confusing since that still mentions `EasyCellularConnection` but that doesn't exist in the sources...

Can you also help me with the two questions above?

AnttiKauppila, 2019-01-07:
@AriParkkila Can you handle this?

AriParkkila, 2019-01-07:
@vidavidorra I guess you could use CellularContext::get_default_instance() and CellularDevice::get_default_instance() like you already found out.

It should make no difference if you connect using `CellularBase` or `CellularContext`. The reason could be that some cellular configuration somehow changes, see defines in _mbed_config.h_ at the BUILD folder.

Can you get debug traces when you query SIM serial number and/or connect to network defining `"cellular.debug-at": true` in your `mbed_app.json`.

It seems that README has some old information and we need to update that, thanks for telling that.

Your sample code seems to call the correct functions to query ME and SIM information. Please be prepared that you need to use `CellularInformation` also for SIM in Mbed OS 5.12.

vidavidorra, 2019-01-08:
@AriParkkila Thanks for the information. I'll try and check the difference between `CellularBase` and `CellularContext`. Which one of those do you suggest using, regarding future development, will both stay in the codebase for a while?

> Can you get debug traces when you query SIM serial number and/or connect to network defining "cellular.debug-at": true in your mbed_app.json.

I'll verify this today and post back the results. As for the file handle argument to `CellularDevice::open_information()`, is that optional in this case, so should it work if I don't pass the file handle I get in my example?

AriParkkila, 2019-01-08:
@vidavidorra `CellularContext` has some extended functionality (e.g. non-IP), but if don't need those you can use `CellularBase`. `CellularBase` is renamed (deprecated) in Mbed OS 5.12 in favor for `CellularInterface`. The both are maintained for now, but `CellularBase` being much simpler is likely also more stable.
You need to give filehandle when calling `CellularDevice::open_information(filehandle)` for now, but that will be optional in Mbed OS 5.12.

vidavidorra, 2019-01-08:
@AriParkkila There is no function available in `CellularBase` and `CellularContext` as far as I know to get the file handle so I had to add the snippet below to `CellularBase` to get the file handle. Is that how it's suppose to go. Could you please advice on how I should get the file handle to pass to `CellularDevice::open_information()`? I forgot to mention that in OP.

```c++
// In CellularDevice::
    FileHandle *get_file_handle() {
        return _fh;
    }
```

AriParkkila, 2019-01-08:
@vidavidorra it seems that Mbed OS 5.11 already has filehandle parameter optional. When calling `open_information` with NULL it uses the device's default filehandle.

vidavidorra, 2019-01-10:
@AriParkkila Apologies for the late reply. I've had problems with my build. Normally I'd build using the release profile, but now needed to build using the debug profile to get those AT debug messages. That  turned out to give a hardfault for some rtx mutex error, but didn't have the time to look into that much further.
Now I've ended up building the release profile but with some debug manually enabled (in the `debug()` functions in `mbed_debug.h`.
Anyhow, these are my logs with the AT commands. And the code that I'm currently using.

## Code
The `TRACE_DEBUG` is for our own logging messages, those are the ones that start with the time. Do note that the code below is called **after** the cellular connection has been made.
This is using the `CellularBase` for the connection and `#define CELLULAR_DEVICE UBLOX_PPP`.
```c++
    CellularDevice *device = CellularDevice::get_default_instance();
    CellularInformation *information = device->open_information();

    if (information == nullptr) {
        TRACE_DEBUG("No information!!!");
    } else {
        char model[9 + 1];
        nsapi_error_t ret = information->get_model(model, sizeof(model));
        device->close_information();
        if (ret != NSAPI_ERROR_OK) {
            TRACE_DEBUG("Cellular model not found (%d)!", ret);
        } else {
            TRACE_DEBUG("Cellular model is: '%s'", model);
        }
    }
    device->close_information();
```

## Log
I'm now trying to get the model in this code because that is one that is already printed in the AT commands before, so that should work. The `get_model` line in the log is from a print that I've added in `AT_CellularInformation::get_model()`.

It seems to me like it's trying to send the correct message in both cases but the second one (for the cellular information just halts for a long time and then results in the error code at the end of the log.
```
get_model
AT+CGMM
```

```
1970-01-01 00:00:00 [3] Creating connection thread (submodules/mbed-system/source/wtg_service/wtg_service.cpp:162)
[0]AT
AT
[252]ATE0
ATE0

OK
AT+CMEE=1

OK
get_model
AT+CGMM

SARA-G350

OK
AT+CGEREP=1

OK
1970-01-01 00:00:03 [3] Status: 0 (4096) (submodules/mbed-system/source/connection/network_interface.cpp:121)
AT+CSQ

+CSQ: 99,99

OK
AT+CPIN?

+CPIN: READY

OK
1970-01-01 00:00:03 [3] Status: 0 (4097) (submodules/mbed-system/source/connection/network_interface.cpp:121)
AT+CGREG=2

OK
AT+CREG=2

OK
AT+CGACT?

OK
AT+CSQ

+CSQ: 99,99

OK
AT+CGREG?

+CGREG: 2,2

OK
AT+CREG?

+CREG: 2,0

OK
AT+COPS?

+COPS: 0

OK
AT+CSQ

+CSQ: 99,99

OK
AT+CGREG?

+CGREG: 2,2

OK
AT+CREG?

+CREG: 2,0

OK
AT+CSQ

+CSQ: 99,99

OK
AT+CGREG?

+CGREG: 2,2

OK
AT+CREG?

+CREG: 2,0

OK

+CREG: 5,"00DE","88B0"

+CGREG: 5,"001970-01-01 00:00:15 [3] Status: 0 (4098) (submodules/mbed-system/source/connection/network_interface.cpp:121)
1970-01-01 00:00:15 [3] Status: 0 (4100) (submodules/mbed-system/source/connection/network_interface.cpp:121)
DE","88B0"
AT+CSQ

+CSQ: 19,0

OK
AT+CGATT?

+CGATT: 1

OK
1970-01-01 00:00:15 [3] Status: 0 (4102) (submodules/mbed-system/source/connection/network_interface.cpp:121)
1970-01-01 00:00:15 [3] Status: 2 (0) (submodules/mbed-system/source/connection/network_interface.cpp:121)
AT+CGDCONT?

OK
AT+CGDCONT=1,"IP","APN_NAME_REMOVED!!!"

OK
AT+CGDATA="PPP",1

CONNECT
1970-01-01 00:00:20 [3] Status: 1 (0) (submodules/mbed-system/source/connection/network_interface.cpp:121)
1970-01-01 00:00:20 [3] Connected! (submodules/mbed-system/source/connection/network_interface.cpp:38)
get_model
AT+CGMM
1970-01-01 00:01:21 [3] Cellular model not found (-3012)! (submodules/mbed-system/source/connection/network_interface.cpp:67)
1970-01-01 00:01:21 [3] Connected to network as 10.9.114.218 (submodules/mbed-system/source/connection/connection.cpp:184)
2019-01-10 07:16:24 [3] Time synced (submodules/mbed-system/source/connection/connection.cpp:204)
```

jarvte, 2019-01-10:
After data mode is entered it's not possible to send at commands.
AT+CGDATA="PPP",1 enters the data state. You should get model before that.
In the beginning of the log model is shown fine.

vidavidorra, 2019-01-10:
@jarvte Thanks for your reply. That makes a whole lot of sense!

How am I supposed to power on the module myself then? What I'm doing at the moment is calling `CellularBase::connect()` to make the connection and the snippet from my previous comment after that. To my knowledge, the `connect` call takes care of powering on the modem, so before calling that, the modem is actually powered off. While it is powered off, I can't get the cellular information. 

Are there functions that I can call that only power on the modem, so I can do the following.
1. Power on the modem
2. Get the device information I need
3. Call `connect` to make the connection.

AriParkkila, 2019-01-11:
@vidavidorra you could call e.g. `CellularContext::set_sim_ready()` first, then query device and SIM information, and finally call `CellularContext::connect()`.

You could try to disable automatic APN lookup, that might improve connecting to a network:
```
            "ppp-cell-iface.apn-lookup": false,
            "cellular.use-apn-lookup": false,
```

vidavidorra, 2019-01-11:
Okay, finally got it working. Thanks for your support guys @AriParkkila @jarvte !

IMHO it would be very useful for other if the following two point are added to the docs.
1. Once a cellular modem goes to the data state (`AT+CGDATA="PPP",1`) it is not possible to send AT commands to the modem.
2. Example/information on how to get cellular information.

## Code
In this code I've used the following interfaces. `_connection` is the `CellularContext` instance and `_interface` is a generic `NetworkInterface`, which simply points to the `CellularContext` instance.
```c++
    CellularDevice *device = CellularDevice::get_default_instance();
    TRACE_DEBUG("device->set_sim_ready() > %d", _connection->set_sim_ready());
    CellularInformation *information = device->open_information();

    if (information == nullptr) {
        TRACE_DEBUG("No information!!!");
    } else {
        char model[9 + 1];
        nsapi_error_t ret = information->get_model(model, sizeof(model));
        device->close_information();
        if (ret != NSAPI_ERROR_OK) {
            TRACE_DEBUG("Cellular model not found (%d)!", ret);
        } else {
            TRACE_DEBUG("Cellular model is: '%s'", model);
        }
    }
    device->close_information();

    CellularSIM *sim = device->open_sim();
    if (sim == nullptr) {
        TRACE_DEBUG("No SIM!!!");
    } else {
        char iccid[100];
        char imsi[100];
        nsapi_error_t imsi_ret = sim->get_imsi(imsi);
        TRACE_DEBUG("SIM IMSI (%d): '%s'", imsi_ret, imsi);
        nsapi_error_t iccid_ret = sim->get_iccid(iccid, 99);
        TRACE_DEBUG("SIM ICCID (%d): '%s'", iccid_ret, iccid);
    }
    device->close_sim();

    TRACE_DEBUG("_interface->connect()");
    auto connect = _interface->connect();
```

vidavidorra, 2019-01-11:
I'll close this now as my issue has been resolved :)

closed 2019-01-11 by vidavidorra