ARMmbed/mbed-os #9275: TCPSocket sigio callback does not specify the event type (LWIP stack)

embeddedteam103, 2019-01-07:
### Description

I'm trying to detect when a socket closes using the `sigio` callback, but unfortunately the `LWIP` event 
type is straight up ignored by mbed.
As there's no `poll` for sockets i basically have no way to know whether the connection is active (except for trying to send data which is not ideal).
Is it on purpose? if it is then what is motive?

Thanks in advance.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

kjbracey-arm, 2019-01-07:
This is on purpose - many stacks don't give any useful information. And it's consistent with normal Unix/POSIX behaviour. Plus it means that socket events don't have to be queued - they contain no information, so they are squishable "flags" (per-socket).

You are expected to find out that a socket has been closed by the return value from a `send` or `recv` - the read or write call will indicate via an error code, or a 0 return for clean end-of-stream on TCP read.

So no special handling for "close events" because they don't exist - just incorporate it into your return value handling. (And if you're not trying to send or recv, you don't need to know the socket's been closed, right?)

kjbracey-arm, 2019-01-07:
It's also consistent with the API being "blocking/non-blocking" rather than "synchronous/asynchronous". You can entirely drive the system using blocking calls, and looking at their return value. You don't need to have an event handler to pick up any information - it's all in the return values.

TacoGrandeTX, 2019-03-14:
@embeddedteam103 We are closing this is a non-issue as explained by @kjbracey-arm

closed 2019-03-14 by TacoGrandeTX