ARMmbed/mbed-os #9276: nrf52: after disconnect, device doesn't go into sleep again

bearsh, 2019-01-07:
### Description

BLE_HeartRate example, release profile, advertising interval changed to 250ms: during advertising, the current consumption is about 76uA. Once connected the controller keeps alive (I don't know if that's normal) and draws a little more then 3.3mA. But once disconnected, it still draws 3.3mA.

Target: nrf52
Toolchain: arm-none-eabi-gcc v7.3.1
Tools: mbed-cli v1.8.3
Mbed OS: c966348d3 (HEAD, tag: mbed-os-5.11.1, origin/mbed-os-5.11) Merge pull request #9208 from ARMmbed/release-candidate

[image]

1. 0s - Blue line (~0.9s): advertising
2. Blue line (~0.9s) - Red line (~4.6s): connected 
3. Red line (~4.6s) - ...: advertising

### Issue request type

- [ ] Question
- [ ] Enhancement
- [x] Bug

bearsh, 2019-01-08:
The same also happens if I set a advertising timeout. Once the timeout is reached, advertising stop but current consumption increases dramatically.

TacoGrandeTX, 2019-01-11:
@bearsh Thanks for pointing this out and we noticed it ourselves while investigating https://github.com/ARMmbed/mbed-os/issues/9093.  We have shown that this is related to the CryptoCell 310 (CC310), and @pan-  has an idea to resolve it.  The higher current floor we observe related to using the CC310 (mentioned in https://github.com/ARMmbed/mbed-os/issues/9093) needs to be re-evaluated.

bearsh, 2019-01-14:
thanks, applying the workaround from https://github.com/ARMmbed/mbed-os/issues/9093#issuecomment-453690139 reduces the current consumption to the desired value.

TacoGrandeTX, 2019-02-06:
@bearsh Are you content and can we close the issue?

bearsh, 2019-02-11:
@TacoGrandeTX I didn't do any testing on the latest release but once #9372 get merged the issue will be closed automatically, so I don't see any reason to close it. furthermore, as far as I got it, there's no proof that #9372 will really solve this issue, even with #9019 merged...

PR #9372 (2019-02-12): BLE - Nordic: Release crypto cell when not in use.
### Description

Previously, the CryptoToolbox was allocated once as part of the security manager.
This was inneficient memory wise as it is only use to prepare key at initialization
and when we need to compute shared keys.
This was also inneficient power consumption wise as the Crypto cell was kept enabled even
when it wasn't used.

This fix creates a CryptoToolbox whenever it is needed and release it once it has fulfilled its
purpose. Note that CryptoToolbox allocation happens on the heap as mbed tls data structure are huge
and there's an hi [... cut]
closed 2019-02-12 by cmonr