ARMmbed/mbed-os #9281: lfs_emubd.c needs to be ignored?

Alex-EEE, 2019-01-07:
### Description

mbed compile tries to build "lfs_emubd.c," which I think should only apply to PC unit tests?  Should this be added to a .mbedignore file?  From my log:

Compile [ 81.9%]: lfs_emubd.c
[Fatal Error] lfs_emubd.c@14,0: [Pe1696]: cannot open source file "dirent.h"

This might slip through building with ARM, but I don't think IAR has this h file.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

closed 2019-01-22 by 0xc0170

davidsaada, 2019-01-24:
@Alex-EEE Can you reopen this issue? @geky has drawn my attention to the fact that these files are already under .medignore in an upper folder. This means that the problem shouldn't exist even before my fix (and that I should revert it as it's redundant). Do you have the full LittleFS directory tree with .mbedignore under littlefs folder?

Alex-EEE, 2019-01-24:
This is probably related to https://github.com/ARMmbed/mbed-os/issues/9419 then.  .mbedignore files are not really working properly in mbed 11.  @bridadan confirmed this bug.

Alex-EEE, 2019-01-24:
I don't think I can re open this since @0xc0170 closed it.

bridadan, 2019-01-24:
@Alex-EEE 

> .mbedignore files are not really working properly in mbed 11. @bridadan confirmed this bug.

Just to be clear, .mbedignores _not under a TESTS directory_ that have an entry for a test case wil not actually ignore the test. The .mbedignore file needs to be inside the TESTS directory in order for it to work.

Alex-EEE, 2019-01-24:
Right.  This wasn't the behavior in mbed 9.  .mbedignore files could be anywhere.  We leverage this capability...we think it's better practice to just have one mbedignore file in root, that way you can easily see in one place all the TESTS being skipped.  Also, we swap the file based on whether we're doing a quick or long CI test.

davidsaada, 2019-01-24:
@bridadan  Thanks for the clarification. Will revert #9440 then.

PR #9440 (2019-01-22): Put LittleFS emulated BD module under .mbedignore
### Description
Put LittleFS emulated BD module under .mbedignore. The emulated BD is required for simulation of a block device outside mbed-os, but not required in mbed-os.

Resolves #9281.

### Pull request type
    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers
@geky