ARMmbed/mbed-os #9282: Getting "COM14 CMD timeout: ifup" when trying to run netsocket tests on DISCO_L475VG_IOT01A board through icetea

Vanireddy6, 2019-01-08:
**Description:**

- Type: Bug
- Priority: Major

**TEST CASES FAILED**:

1. SOCKET_BIND_PORT
2.TCPSERVER_ACCEPT
3.TCPSOCKET_ACCEPT
4.TCPSOCKET_ECHOTEST_BURST_SHORT

**STEPS TO REPRODUCE**:

1. Imported "mbed-os-example-sockets" example and downloaded "wifi-ism43362" drivers using "mbed add" comand.
2.Added WiFi credentials (SSID & PASSWORD) in "ISM43362Interface.JSON" and "HeapBlockDeviceAndWifiInterface.json" files.
3.Run using "mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n TCPSOCKET_ECHOTEST_BURST_SHORT".

**Logs:**
C:\Users\vann\mbed\mbed-os-example-sockets\mbed-os>mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n TCPSOCKET_ECHOTEST_BURST_SHORT --test-config tools/test_configs/ISM43362Interface.JSON
[mbed] Working path "C:\Users\vann\mbed\mbed-os-example-sockets\mbed-os" (library)
[mbed] Program path "C:\Users\vann\mbed\mbed-os-example-sockets\mbed-os"
[ERROR] Attempt to override undefined parameter 'ism43362.provide-default' in 'application[*]'
cleaning up Icetea log directory.
12:25:51.334 Using Icetea version 1.0.2
12:25:51.334 Loading suite from file
12:25:51.334 Importing testcases for filtering
12:25:51.411 Filtering testcases
12:25:51.411 Suite creation complete.
12:25:51.415 Test cases prepared.
12:25:51.420 Starting suite.
12:25:51.420 Starting repeat 1 of 1
12:25:51.420 Iteration 1 of 1
12:25:51.427
12:25:51.431 START TEST CASE EXECUTION: 'TCPSOCKET_ECHOTEST_BURST_SHORT'
12:25:51.431
12:25:51.476 | TC     MainThread: Start Test case 'TCPSOCKET_ECHOTEST_BURST_SHORT'                                      12:25:51.476 | TC     MainThread: ====setUpTestBench====
12:25:51.489 | TC     MainThread: ====setUpTestBench====
12:25:51.493 | TC     MainThread: ---------------------
12:25:51.493 | TC     MainThread: Initialize DUT's connections                                                          12:25:51.493 | RSP     MainThread: Allocating 1 duts.
12:25:51.530 | LAL     MainThread: Allocated device 0764022118486110270CF749
12:25:51.530 | RSP     MainThread: Allocation successful
12:25:51.542 | D1     Thread-3: Flashing dev: 0764022118486110270CF749
12:25:51.578 | FLS     Thread-3: sendBreak to device to reboot
12:25:51.829 | FLS     Thread-3: reset completed
12:26:02.986 | FLS     Thread-3: sendBreak to device to reboot
12:26:03.243 | FLS     Thread-3: reset completed
12:26:03.645 | FLS     Thread-3: 0764022118486110270CF749 flash success
12:26:03.645 | TC     Thread-3: flash ready
12:26:03.645 | TC     Thread-3: Use board DISCO_L475VG_IOT01A as "D1" (id: 0764022118486110270CF749)
12:26:03.707 | D1     MainThread: DutSerial COM14

+----------+--------------+
| COM port |   Location   |
+----------+--------------+
|  COM14   | X = 0, Y = 0 |
+----------+--------------+
12:26:03.714 | D1 <-> MainThread: Open Connection for 'D1' using 'COM14' baudrate: 115200
12:26:03.726 | D1 <-> MainThread: Use normal serial write mode
12:26:03.726 | TC     MainThread: Waiting for dut 1 to initialize.
12:26:03.730 | D1 --> DutThread: set --retcode true
/>set --retcode true  DutThread: />s
12:26:03.786 | D1 <-- DutThread: retcode: 0
12:26:03.802 | D1 --> DutThread: echo off
/>echo off 2 | D1 <-- DutThread: />
12:26:03.835 | D1 <-- DutThread: ECHO is off
12:26:03.846 | D1 <-- DutThread: retcode: 0
12:26:03.878 | D1 --> DutThread: set --vt100 off
12:26:03.894 | D1 <-- DutThread: retcode: 0
12:26:03.903 | D1 --> DutThread: set dut "COM14"
12:26:03.918 | D1 <-- DutThread: retcode: 0
12:26:03.927 | D1 --> DutThread: set testcase ""
12:26:03.934 | TC     MainThread: ------TC SET UP---------
12:26:03.943 | D1 <-- DutThread: retcode: 0
12:26:03.950 | D1 --> DutThread: ifup
12:26:03.979 | D1 <-- DutThread: ++ MbedOS Fault HanHardF
12:26:04.003 | D1 <-- DutThread: Con
12:26:04.016 | D1 <-- DutThread: R0   :
12:26:04.028 | D1 <-- DutThread: R1   :
12:26:04.039 | D1 <-- DutThread: R2   :
12:26:04.052 | D1 <-- DutThread: R3   :
12:26:04.063 | D1 <-- DutThread: R4   :
12:26:04.075 | D1 <-- DutThread: R5   :
12:26:04.088 | D1 <-- DutThread: R6   :
12:26:04.099 | D1 <-- DutThread: R7   :
12:26:04.111 | D1 <-- DutThread: R8   :
12:26:04.125 | D1 <-- DutThread: R9   :
12:26:04.137 | D1 <-- DutThread: R10  :
12:26:04.148 | D1 <-- DutThread: R11  :
12:26:04.160 | D1 <-- DutThread: R12  :
12:26:04.173 | D1 <-- DutThread: SP   : 20003B50
12:26:04.184 | D1 <-- DutThread: LR   : 0801DCC7
12:26:04.197 | D1 <-- DutThread: PC   : 0800031E
12:26:04.209 | D1 <-- DutThread: CPUID:
12:26:04.220 | D1 <-- DutThread: HFSR : 40000000
12:26:04.233 | D1 <-- DutThread: MMFSR: 00000000
12:26:04.244 | D1 <-- DutThread: BFSR : 00000082
12:26:04.256 | D1 <-- DutThread: BFAR :
12:26:04.269 | D1 <-- DutThread: Mode :
12:26:04.280 | D1 <-- DutThread: Priv : P
12:26:04.292 | D1 <-- DutThread: Stack
12:26:04.316 | D1 <-- DutThread: -- MbedOS Fau
12:26:04.341 | D1 <-- DutThread: ++ MbedOS Error Info ++
12:26:04.352 | D1 <-- DutThread: Error Status: 0x80Fault ex
12:26:04.365 | D1 <-- DutThread: Location:
12:26:04.377 | D1 <-- DutThread: Error Value
12:26:04.388 | D1 <-- DutThread: Current Thread: main  Id: 0x20002BD4 Entry: 0x80210DF
12:26:04.401 | D1 <-- DutThread: For more info, visit: https://armmbed.g
12:26:04.413 | D1 <-- DutThread: -- MbedOS E
12:26:04.424 | D1 <-- DutThread: == The system has been rebooted due to a fatal error. ==
12:26:04.437 | D1 <-- DutThread: ARM Ltd
12:26:05.640 | D1 <-- DutThread: />
12:26:54.017 | D1     MainThread: CMD timeout: ifup
12:26:54.022 | TC     MainThread: Test Case fails because of: COM14 CMD timeout: ifup
12:26:54.026 | TC     MainThread: ====tearDownTestBench====
12:27:45.150 | D1     MainThread: CMD timeout: ifup
12:28:36.349 | D1     MainThread: CMD timeout: ifup
12:28:36.796 | TC     MainThread: Test 'TCPSOCKET_ECHOTEST_BURST_SHORT' FAIL, reason: COM14 CMD timeout: ifup
12:28:37.575 Test case TCPSOCKET_ECHOTEST_BURST_SHORT failed, No retries left.

+--------------------------------+---------+-------------------------+-------------+---------------------+----------+
| Testcase                       | Verdict |       Fail Reason       | Skip Reason |      platforms      | duration |
+--------------------------------+---------+-------------------------+-------------+---------------------+----------+
| TCPSOCKET_ECHOTEST_BURST_SHORT |   fail  | COM14 CMD timeout: ifup |             | DISCO_L475VG_IOT01A | 165.372  |
+--------------------------------+---------+-------------------------+-------------+---------------------+----------+
+---------------+----------------+
|    Summary    |                |
+---------------+----------------+
| Final Verdict |      FAIL      |
|     count     |       1        |
|    passrate   |     0.00 %     |
|      fail     |       1        |
|    Duration   | 0:02:45.372000 |
+---------------+----------------+
12:28:37.644 Cleanup done.

0xc0170, 2019-01-15:
Similar to another issue related also to ISM43362, This might be better place in ISM43362 driver repository.

@ARMmbed/team-st-mcd Please review (and we can move this one to the drivers repository)

jeromecoutant, 2019-01-15:
Hi
Your log begins with:

[ERROR] Attempt to override undefined parameter 'ism43362.provide-default' in 'application[*]'
cleaning up Icetea log directory.

Can you check this first ?

Vanireddy6, 2019-01-17:
Hi
Could you please let me know how do check that?

jeromecoutant, 2019-02-20:
Did you add 
"ism43362.provide-default": true
in your config ?

jeromecoutant, 2019-06-17:
I think we can close the issue ?
Thx

closed 2019-08-14 by 0xc0170