ARMmbed/mbed-os #9285: Why is SDBlockDeviceDriver in a component

janjongboom, 2019-01-08:
### Description

As far as I can see the SDBlockDeviceDriver does not have any overhead when not used. Why is this in a component then? This makes discoverability much harder, as even I need to look up the components_add syntax. See also the comments in https://os.mbed.com/questions/82718/SDBlockDevice-does-not-work-with-mbed-51/?c=31992.

I think this was not the case for Mbed OS 5.10. 

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

loverdeg-ep, 2019-01-08:
I still wonder what the component concept is implying.

optional software component?
driver for external IC component?

janjongboom, 2019-04-24:
@cmonr @bulislaw Why is this closed?

closed 2019-10-31 by ciarmcom