ARMmbed/mbed-os #9298: Why there are two different AT command components ?

prowan, 2019-01-09:
### Description
Since both of them are dealing with AT commands, why ATHandler and ATCmdParser these two different components are set up ? What is the difference between them? Can they replace each other?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

prowan, 2019-01-09:
I am trying to pen this link https://jira.arm.com/browse/MBOCUSTRIA-406 
but I got 
[image]

Is there any other style to open the link ? Thanks a lot !

0xc0170, 2019-01-09:
It's internal ticket reference.

cc @ARMmbed/mbed-os-wan

mirelachirica, 2019-01-10:
ATCmdParser cannot be used with current AT cellular interfaces under /features/cellular. They are all inheriting AT_CellularBase which only supports ATHandler.

ATHandler is optimized for simpler syntax for parsing more complex, multi(optional)line, optional parameters or some of the vendor specific special cases of AT responses that we bumped into during cellular development. It works also with much smaller internal buffer of 16bytes compare to 256bytes default buffer size for ATCmdParser. But simpler parsing for complex AT responses was, from my point of view, the main rational behind ATHandler.

blind-owl, 2019-01-10:
This issue is managed here:
https://os.mbed.com/questions/84014/How-to-choose-between-ATHandler-and-ATCm/

prowan, 2019-01-11:
Your reply gave me the confidence to continue studying and researching mbed, thank you for your guidance.

closed 2019-01-14 by ciarmcom