ARMmbed/mbed-os #9299: How to write and execute testcases for ble apps through icetea in mbed?

Vanireddy6, 2019-01-09:
I have started  testing Mbed BLE applications like Ble_Heartrate,Battery service...
I have to write testcases for BLE API's and test through ICETEA, but i am not finding any example to refer.
Could you please let me know , how to write testcases and run icetea on that?

cmonr, 2019-01-17:
@ARMmbed/mbed-os-test-team or @ARMmbed/mbed-os-pan Thoughts?

OPpuolitaival, 2019-01-17:
@Vanireddy6 documentation for icetea can be found https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/tools/testing/icetea.md

OPpuolitaival, 2019-01-17:
You might be interested also an example how to write test where two mbed devices are connecting to each other during the test: https://github.com/ARMmbed/mbed-os/blob/master/TEST_APPS/testcases/nanostack_mac_tester/send_data.py

Vanireddy6, 2019-01-21:
Hi OPpuolitaival,
Thanks for  providing that link.
Actually i want to write a testcase to call the BLE API's....Is there any example present for that?
Please let me know if it is..

0xc0170, 2019-08-14:
> Actually i want to write a testcase to call the BLE API's....Is there any example present for that?

@ARMmbed/mbed-os-pan I don't think there is any is there?

@Vanireddy6 Have you found your answer meanwhile.

closed 2019-10-24 by 0xc0170