ARMmbed/mbed-os #9303: rtos: ThisThread::flag_wait_all() doesn't work with multiple flags

hasnainvirk, 2019-01-09:
### Description

ThisThread::flag_wait_all() should work with any number of flags specified. 
However, if you are looking for multiple flags, it doesn't work. Here is the example code

```C
#define LED_ON_FLAG     0x01
#define LED_OFF_FLAG    0x02

// Thread object
Thread thread;

// DitigtalOut object (LED)
DigitalOut led(LED1);

void thread_task() {
    
    for(;;) {
        // Thread should wait for the event flags determined here.
        // Any event flags will automatically be cleared.
        uint32_t flags = ThisThread::flags_wait_all(LED_ON_FLAG | LED_OFF_FLAG);
        
        if (flags & LED_ON_FLAG) {
            led = 1;
        }
        
        if (flags & LED_OFF_FLAG) {
            led = 0;
        } 
    }
}

int main (void) {
    uint32_t flag = LED_ON_FLAG;
    thread.start(callback(thread_task));

    while (true) {
        // wait for 1 second
        wait_ms(1000);
        
        // set the flag to the Thread
       uint32_t ret = thread.flags_set(flag);
        
        // Alternate between flags for the next run
        if (flag == LED_ON_FLAG) {
            flag = LED_OFF_FLAG;
        } else {
            flag = LED_ON_FLAG;
        }
    }
}
````

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X ] Bug

hasnainvirk, 2019-01-09:
@kjbracey-arm Please have a look.

kjbracey-arm, 2019-01-09:
Don't you want `wait_any`? `wait_all` will wait for both flags to be set.

hasnainvirk, 2019-01-09:
Ah, alright, I read the document in wrong way. I thought, wait_all means wait for one of the given signals. Closing this.

closed 2019-01-09 by hasnainvirk

kjbracey-arm, 2019-01-09:
BTW, if you're in a "use `ThisThread`" sort of mood, then `wait_ms` should be `ThisThread::sleep_for(1000)`.