ARMmbed/mbed-os #9314: Mbed OS on FreeRTOS?

bmcdonnell-ionx, 2019-01-09:
### Description

What would be involved in enabling Mbed OS to run on FreeRTOS (replacing RTX5)? Are there any plans to facilitate or support this?

I see that [RTX5 implements](http://www2.keil.com/mdk5/cmsis/rtx) the [CMSIS-RTOS v2 API](http://arm-software.github.io/CMSIS_5/RTOS2/html/index.html). [CMSIS-FreeRTOS](https://github.com/ARM-software/CMSIS-FreeRTOS) also exists as another [implementation of the CMSIS-RTOS API](https://arm-software.github.io/CMSIS-FreeRTOS/General/html/index.html), so Mbed OS could conceivably be made to run on either in a sane way.

I did a quick search in `mbed-os` (`grep -Ri "#include.*rtx" .`), and I see that some things include RTX headers directly. I guess this means that Mbed OS hasn't restricted itself to the CMSIS_RTOS API? i.e. That would be a prerequisite to such a port, right?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-01-10:
This is related and taking it more generally: https://github.com/ARMmbed/mbed-os/issues/9139 where we can continue to keep track (there are few issues related to this still opened in this repo and also in CMSIS_5).

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
I think the question has been answered. Please reopen if you're not satisfied with the answer.

bmcdonnell-ionx, 2019-04-30:
Am I inferring the answers correctly?

- What's required is mainly described in the OP and @0xc0170's comment on #9139.
- Per @0xc0170's comment there, you're interested in getting to using vanilla CMSIS as the interface that Mbed uses to interact w/ the RTOS, but there is no publicly-shared timetable. (Is this on your task list or backlog?)
- No plans yet to implement Mbed on other RTOSes. (Should I log another issue as a feature request, instead of just a question?)
- Further action on these items will be tracked on #9139.

0xc0170, 2019-05-03:
> Per @0xc0170's comment there, you're interested in getting to using vanilla CMSIS as the interface that Mbed uses to interact w/ the RTOS, but there is no publicly-shared timetable. (Is this on your task list or backlog?)

Backlog currently.

> No plans yet to implement Mbed on other RTOSes. (Should I log another issue as a feature request, instead of just a question?)

Not at the moment.

0xc0170, 2019-05-03:
Lets continue looking at #9139