ARMmbed/mbed-os #9322: Doxygen CI check should report warnings, not only errors

0xc0170, 2019-01-10:
### Description

See this job: https://travis-ci.org/ARMmbed/mbed-os/jobs/477345635 . There are lot of warnings that should not be there!

I would propose the way we did astyle, do it in few steps:

1. warning number in the pull request reported (would be now around 160)
2. clean-up to get to 0, document that doxygen should not generate any warning
3.  warning -> error

cc @melwee01 @AnotherButler can you help with these warnings?

Thanks to @SeppoTakalo for the report:

```
$ doxygen doxyfile_options 2>&1 |wc -l
160
```

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2019-01-10:
> warning number in the pull request reported (would be now around 160)

cc @ARMmbed/mbed-os-test

OPpuolitaival, 2019-01-10:
doxygen doxyfile_options 2>&1 | wc -l 
257

@0xc0170 what we should do?

0xc0170, 2019-01-10:
I would propose to report the number in PR (similar to what we had in astyle)

Start tracking warning number , no more new doxy warnings, so if we can get the number in each PR (+/-) plus also total (until we get to 0).

`travis-ci/docs Passed, 257 total warnings, +/- 0 in this PR" ?

AnotherButler, 2019-01-11:
@0xc0170 What type of help would you like?

0xc0170, 2019-01-11:
> @0xc0170 What type of help would you like?

To clean up all warnings for Mbed OS master

AnotherButler, 2019-01-14:
By clean up, do you mean clarify the language?

0xc0170, 2019-01-15:
> doxygen doxyfile_options 2>&1 |wc -l

This to report 0 - no warnings from doxygen run

SeppoTakalo, 2019-01-15:
Oh.. hold, my 160 is AFTER the 2 day write-a-thon event. On a branch where we fixed most of Cellular and Socket problems. Some LoRa as well.

So both numbers might be correct, it just depends in which branch.

And our fixes are not yet merged in https://github.com/ARMmbed/mbed-os/pull/9328

SeppoTakalo, 2019-01-15:
Also, documentation team is a wrong entity to clean up engineering problems.

Somebody should go on and look at the warnings reported, then assign somebody from each team to fix all warnings regarding their area.

```
/Users/septak01/src/mbed-os/components/TARGET_PSA/services/crypto/COMPONENT_PSA_SRV_IPC/crypto_platform_spe.h:93: warning: group PSA: ignoring title "Crypto structures for IPC" that does not match old title "Crypto APIs"

/Users/septak01/src/mbed-os/components/TARGET_PSA/services/crypto/COMPONENT_PSA_SRV_IPC/crypto_platform_spe.h:2: warning: the name `psa/crypto_platform_spe.h' supplied as the second argument in the \file statement is not an input file
/Users/septak01/src/mbed-os/components/TARGET_PSA/services/crypto/COMPONENT_PSA_SRV_IPC/crypto_struct_ipc.h:4: warning: the name `psa/crypto_struct.h' supplied as the second argument in the \file statement is not an input file
/Users/septak01/src/mbed-os/components/TARGET_PSA/services/crypto/COMPONENT_SPE/crypto_spe.h:2: warning: the name `psa/crypto_spe.h' supplied as the second argument in the \file statement is not an input file
/Users/septak01/src/mbed-os/features/nfc/stack/tech/iso7816/iso7816_app.h:37: warning: Compound nfc_tech_iso7816_app is not documented.
/Users/septak01/src/mbed-os/features/lorawan/lorastack/phy/LoRaPHYKR920.h:50: warning: Compound LoRaPHYKR920 is not documented.
/Users/septak01/src/mbed-os/features/netsocket/DTLSSocket.h:28: warning: Compound DTLSSocket is not documented.
/Users/septak01/src/mbed-os/components/TARGET_PSA/services/crypto/COMPONENT_PSA_SRV_IPC/crypto_struct_ipc.h:42: warning: Compound psa_aead_operation_s is not documented.
/Users/septak01/src/mbed-os/features/storage/filesystem/fat/ChaN/ff.h:138: warning: Compound FFOBJID is not documented.
/Users/septak01/src/mbed-os/features/netsocket/WiFiInterface.h:29: warning: Compound WiFiInterface is not documented.
/Users/septak01/src/mbed-os/features/lorawan/system/lorawan_data_structures.h:1110: warning: Compound loramac_protocol_params is not documented.
/Users/septak01/src/mbed-os/features/cellular/framework/API/CellularNetwork.h:144: warning: Compound mbed::CellularNetwork::operator_names_t is not documented.
/Users/septak01/src/mbed-os/platform/CircularBuffer.h:40: warning: Compound mbed::internal::is_unsigned< unsigned int > is not documented.
/Users/septak01/src/mbed-os/features/netsocket/TCPServer.h:31: warning: Compound TCPServer is not documented.
/Users/septak01/src/mbed-os/features/nfc/stack/tech/type4/type4_target.h:40: warning: Compound nfc_tech_type4_target is not documented.
/Users/septak01/src/mbed-os/features/nfc/stack/transceiver/transceiver.h:119: warning: Compound __nfc_b_info is not documented.
/Users/septak01/src/mbed-os/components/TARGET_PSA/services/crypto/COMPONENT_PSA_SRV_IPC/crypto_struct_ipc.h:46: warning: Compound psa_crypto_generator_s is not documented.
/Users/septak01/src/mbed-os/TESTS/netsocket/dns/dns_tests.h:45: warning: Compound dns_application_data is not documented.
/Users/septak01/src/mbed-os/features/nfc/acore/acore/ac_stream.h:52: warning: Compound __ac_ostream is not documented.
/Users/septak01/src/mbed-os/features/storage/filesystem/littlefs/littlefs/lfs.h:178: warning: Compound lfs_info is not documented.
/Users/septak01/src/mbed-os/components/TARGET_PSA/spm/COMPONENT_SPE/spm_internal.h:132: warning: Compound spm_partition is not documented.
/Users/septak01/src/mbed-os/platform/CircularBuffer.h:44: warning: Compound mbed::internal::is_unsigned< unsigned long > is not documented.
/Users/septak01/src/mbed-os/platform/CircularBuffer.h:32: warning: Compound mbed::internal::is_unsigned< unsigned char > is not documented.
/Users/septak01/src/mbed-os/features/lorawan/lorastack/phy/LoRaPHYAU915.h:52: warning: Compound LoRaPHYAU915 is not documented.
/Users/septak01/src/mbed-os/TESTS/mbed_drivers/timeout/timeout_tests.h:340: warning: Compound TimeoutDriftTester is not documented.
/Users/septak01/src/mbed-os/platform/mbed_retarget.h:506: warning: Compound pollfd is not documented.
/Users/septak01/src/mbed-os/features/storage/filesystem/littlefs/littlefs/lfs.h:271: warning: Compound lfs is not documented.
/Users/septak01/src/mbed-os/features/nanostack/mbed-mesh-api/mbed-mesh-api/MeshInterfaceNanostack.h:155: warning: Compound MeshInterfaceNanostack is not documented.
/Users/septak01/src/mbed-os/features/lorawan/lorastack/phy/LoRaPHYUS915.h:50: warning: Compound LoRaPHYUS915 is not documented.
/Users/septak01/src/mbed-os/components/TARGET_PSA/spm/COMPONENT_SPE/spm_internal.h:123: warning: Compound spm_active_msg is not documented.
/Users/septak01/src/mbed-os/features/netsocket/NetworkStack.h:37: warning: Compound NetworkStack is not documented.
/Users/septak01/src/mbed-os/hal/crc_api.h:40: warning: Compound crc_mbed_config is not documented.
/Users/septak01/src/mbed-os/features/storage/filesystem/littlefs/littlefs/lfs.h:215: warning: Compound lfs_file is not documented.
/Users/septak01/src/mbed-os/features/cellular/framework/AT/ATHandler.h:57: warning: Compound mbed::device_err_t is not documented.
/Users/septak01/src/mbed-os/events/equeue/equeue.h:69: warning: Compound equeue::equeue_slab is not documented.
/Users/septak01/src/mbed-os/features/storage/filesystem/littlefs/littlefs/lfs.h:171: warning: Compound lfs_file_config is not documented.
/Users/septak01/src/mbed-os/components/TARGET_PSA/spm/COMPONENT_SPE/handles_manager.h:81: warning: Compound psa_handle_manager_t is not documented.
[... cut ...]
/Users/septak01/src/mbed-os/platform/mbed_error.h:959: warning: The following parameters of mbed_error_reboot_callback(mbed_error_ctx *error_context) are not documented:
  parameter 'error_context'
/Users/septak01/src/mbed-os/features/netsocket/TLSSocketWrapper.h:187: warning: Found unknown command `\parameter'
/Users/septak01/src/mbed-os/features/netsocket/TLSSocketWrapper.h:187: warning: Found unknown command `\parameter'
/Users/septak01/src/mbed-os/features/netsocket/TLSSocketWrapper.h:187: warning: Found unknown command `\parameter'
/Users/septak01/src/mbed-os/features/netsocket/TLSSocketWrapper.h:187: warning: Found unknown command `\parameter'
```

SeppoTakalo, 2019-01-15:
Easy to spot teams from there.

closed 2019-10-31 by ciarmcom