ARMmbed/mbed-os #9353: Documentation: Incorrect label on Nucleo-144 graphical pinout

jorisa, 2019-01-11:
### Description

PC_11 on CN11 should have been CN1 Nucleo-144 platform (for example [ST-Nucleo-F767ZI](https://os.mbed.com/platforms/ST-Nucleo-F767ZI/)) from looking at the mb1137 Altium project.

### Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

jorisa, 2019-01-12:
[image]

jeromecoutant, 2019-01-14:
Thx!

Same issue found for NUCLEO_F429ZI, NUCLEO_F746ZG, NUCLEO_F767ZI, and NUCLEO_H743ZI !

jorisa, 2019-01-14:
Thank you! The "pins connected to other peripherals" (darkblue) is new I assume? That's a helpful addition.

jorisa, 2019-01-15:
Reopened the issue, the right side image (CN12) should also be updated. The updated image shows "pins connected to other peripherals", while the "old" one does not. this gives the false impression that non of these pins are in use.

jeromecoutant, 2019-01-15:
I uploaded all the updated images!
Thx

closed 2019-01-17 by cmonr