ARMmbed/mbed-os #9364: Test case failed with "CMD timeout: set --retcode true" , while creating object instance and  running icetea on Mbed-BLE-examples

Vanireddy6, 2019-01-14:
**BOARD USED**: DISCO_L475VG_IOT01A
**Steps to Reproduce**:
1.Import "mbed-os-example-ble-Thermometer" example.
2.Goto the path "mbed-os/TEST-APPS/device/main.cpp" and open main.cpp file 
3.Add "#include ble/BLE.h" and create instance for BLE(BLE &ble = BLE::Instance();), please refer the attached main.cpp file for referrence.
4.Run the command "mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED" and observe the below logs.

**Logs**:

C:\Users\vann\mbed\mbed-os-example-ble-Thermometer\mbed-os>mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED
[mbed] Working path "C:\Users\vann\mbed\mbed-os-example-ble-Thermometer\mbed-os" (library)
[mbed] Program path "C:\Users\vann\mbed\mbed-os-example-ble-Thermometer\mbed-os"
[Warning] Test with name 'test_cmdline' was not found in the available tests
Building library mbed-build (DISCO_L475VG_IOT01A, GCC_ARM)
Scan: mbed-os
Copy: gpio_irq_device.h
Copy: cfg_thread_end_device.h
Copy: pn512.h
Copy: pn512_poll.h
Copy: thread_extension.h
Copy: fnet_netif.h
Copy: stm32l4xx_hal_irda.h
Copy: thread_leader_service.h
Copy: pwmout_api.h
Copy: chap_ms.h
Copy: chap-new.h
Copy: pem.h
Copy: net_6lowpan_parameter_api.h
Copy: mac_filter_api.h
Copy: fnet_mempool.h
Copy: ppp_opts.h
Copy: LoRaMac.h
Copy: stm32l4xx_hal_dfsdm.h
Copy: mbed_version.h
Copy: device.h
Copy: stm32l4xx_hal_sram.h
Copy: thread_dhcpv6_server.h
Copy: arm_hal_nvm.h
Copy: ccm.h
Copy: utest_harness.h
Copy: eap.h
Copy: cfg_ethernet_host.h
Copy: Kernel.h
Copy: core_cm33.h
Copy: QUECTEL_BC95_CellularSIM.h
Copy: NFCRemoteEndpoint.h
Copy: event.h
Copy: stm32l4xx_hal_nand.h
Copy: thread_neighbor_class.h
Copy: QUECTEL_UG96_CellularNetwork.h
Copy: iso7816_defs.h
Copy: stm32l4xx_ll_dma.h
Copy: stm32l4xx_hal_sai.h
Copy: DigitalOut.h
Copy: nvstore.h
Copy: ip.h
Copy: ip4.h
Copy: CellularUtil.h
Copy: timer_sys.h
Copy: stm32l4xx_hal_flash_ramfunc.h
Copy: fnet_services_config.h
Copy: inet.h
Copy: GEMALTO_CINTERION_Module.h
Copy: stm32l4xx_ll_system.h
Copy: stm32l4xx_ll_tim.h
Copy: core_sc000.h
Copy: thread_management_server.h
Copy: entropy_poll.h
Copy: fhss_common.h
Copy: mac_fhss_callbacks.h
[... 1950 lines cut ...]
09:49:10.803 | D1     MainThread: CMD timeout: set --retcode true
09:50:01.918 | D1     MainThread: CMD timeout: set --retcode true
09:50:02.700 | TC     MainThread: Test Case fails because of: COM14 CMD timeout: set --retcode true
09:50:02.700 | TC     MainThread: Test 'test_cmdline' FAIL, reason: COM14 CMD timeout: set --retcode true
09:50:03.434 Test case test_cmdline failed, No retries left.

+--------------+--------------+---------------------------------------+-------------+-----------+----------+
| Testcase     |   Verdict    |              Fail Reason              | Skip Reason | platforms | duration |
+--------------+--------------+---------------------------------------+-------------+-----------+----------+
| test_cmdline | inconclusive | COM14 CMD timeout: set --retcode true |             |           | 158.439  |
+--------------+--------------+---------------------------------------+-------------+-----------+----------+
+---------------+----------------+
|    Summary    |                |
+---------------+----------------+
| Final Verdict |  INCONCLUSIVE  |
|     count     |       1        |
|    passrate   |     0.00 %     |
|  inconclusive |       1        |
|    Duration   | 0:02:38.439000 |
+---------------+----------------+
09:50:03.512 Cleanup done.

Please let me know if any thing wrong...

[main.txt](https://github.com/ARMmbed/mbed-os/files/2754054/main.txt)

0xc0170, 2019-01-14:
@ARMmbed/mbed-os-test Can your review the steps above? Is this supported ?

pan-, 2019-01-14:
The Disco board requires [extra config](https://github.com/ARMmbed/mbed-os-example-ble/blob/8e791bfcb3f2098a911f80e554e90a26d220cb47/BLE_GattClient/mbed_app.json#L11-L14) and an [external driver](https://github.com/ARMmbed/cordio-ble-x-nucleo-idb0xa1) to support BLE.

0xc0170, 2019-01-16:
> The Disco board requires extra config and an external driver to support BLE.

@Vanireddy6 is your issue resolved ?

pan-, 2019-03-05:
@Vanireddy6 Just looked at the file attached:

```c++
int main()
{
	eventQueue.call_every(500, periodicCallback);

    BLE &ble = BLE::Instance();
    ble.onEventsToProcess(scheduleBleEventsProcessing);
    ble.init(bleInitComplete);

    eventQueue.dispatch_forever();

    return 0;
    cmd_init(&wrap_printf);

    int c;
    while ((c = getchar()) != EOF) {
        cmd_char_input(c);
    }
    return 0;
}
```

Your main is not correct, the line `eventQueue.dispatch_forever();` dispatch events from the event queue; forever. It means that the program never read the input to get commands from user. Please use UARTSerial in non blocking mode: you can use [sigio](https://github.com/ARMmbed/mbed-os/blob/355f09bbac5a2459e62477fae15d71a005cdc278/drivers/UARTSerial.h#L202) to defer into the event queue an event that fetches character read.

0xc0170, 2019-10-24:
I'll close this as there has not been any further questions

closed 2019-10-24 by 0xc0170