ARMmbed/mbed-os #9379: Sockets on AT modems are created when app calls sendto/recvfrom

oscarh, 2019-01-15:
### Description

An UDP socket is first created on the AT modem when calling sendto/receivefrom. This seems to be on purpose as I can find the flollowing comment on line 138 in features/cellular/framework/AT/AT_CellularStack.cpp:
> // create local socket structure, socket on modem is created when app calls sendto/recvfrom

I find this problematic, since it's harder to tell what went wrong. In my application, I can "successfully" create a socket. When I call sendto, it actuallly first tries to do a DNS lookup, which fails. The socket is still not created... Sendto fails, but WHY this happens is a bit harder to guess, compared to if I would know that I cannot create a socket. I also try to read from the socket, which tries to create it, which fails.

Unfortunanetlly the `_connection_status` is also still `NSAPI_STATUS_GLOBAL_UP`. My modem is in a broken state... I the application had at least failed at creating the socket, I could guess that I need to repare the modem state.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2019-01-17:
@ARMmbed/mbed-os-wan Please review.

AriParkkila, 2019-01-17:
@oscarh thanks for your question(s).

1. You probably should try to resolve IP address first, and then use that IP address with `sendto`, e.g.:
```
SocketAddress saddr;
ret = netif->gethostbyname("example.com", &saddr);
ret = sendto(saddr, data, size);
```
2. `AT_CellularStack` API need to support many cellular modems, and unfortunately IP stack implementations are very much modem specific. It's quite common that when we create sockets (via AT commands) we must give also remote IP address, and that is usually known only while calling `sendto`.
3. `NSAPI_STATUS_GLOBAL_UP` indicates that the modem has packet data service available for sockets (that is, active PDP context or EPS bearer). You can verify that network status is really correct by calling `AT_CellularNetwork::is_active_context()` and checking from the response that your current context is active.

oscarh, 2019-01-17:
Hi @AriParkkila,

Thanks for the answers.

Regarding 2:
At least I undestand WHY we're creating the socket that late now. It had been great if the comment documented the why, not the fact (which can be found in the source code).

Regarding 3:
I'm still a bit confused that the code returns `NSAPI_STATUS_GLOBAL_UP` when the modem really isn't connected/attached. Should `NSAPI_STATUS_GLOBAL_UP` always be return when a context is created, also when I've lost reception and received for instance `+CGREG: 2`. If not, the `_connect_status` dosen't seem to be properly updated, but I'll try to get back to that later (hopefully with a pull request) in that case.

AriParkkila, 2019-01-17:
@oscarh thanks, good points.

2) We need to update `AT_CellularStack` doxygen with some rationale.

3) When network connection is lost (e.g. `CREG: 2`) cellular sends `NSAPI_STATUS_DISCONNECTED` to indicate that network is not available and also to give a hint that sockets should be reinitialized. You are right that in most cases context is readily active after re-registering to the network. However, I guess it's better to indicate to the application that the network is not available, and also that the PDP context might have changed (e.g. IP address) in which case also sockets may need to be reinitialized.

AriParkkila, 2019-01-22:
@oscarh can we close this issue, and resolve network connectivity issues in #9382 ?

cmonr, 2019-01-23:
@AriParkkila If PR https://github.com/ARMmbed/mbed-os/issues/9382 will resolve this issue, then current convention is to keep the issue open until the PR is merged and confirmed to fix the issue.

cmonr, 2019-01-23:
> Or am I missing something?

Yes, yes I am. That's not a PR, it's an issue...
Carry on.

AriParkkila, 2019-04-16:
@oscarh can we close this issue? The latest Mbed OS master with PR #10402 now passes `mbed test -n tests-network-interface --app-config=tools/test_configs/CellularInterface.json` in both PPP and AT mode.

0xc0170, 2019-04-16:
Closing as fixed

closed 2019-04-16 by 0xc0170