ARMmbed/mbed-os #9382: Network status callbacks and the NetworkInterface state machine for AT Cellular modems

oscarh, 2019-01-15:
### Description

The network connectivity states described  [here](https://os.mbed.com/docs/mbed-os/v5.11/apis/network-interfaces.html) claims that the only way to transition from `Global route found` to `Disconnected` is by calling `NetworkInterface::disconnect()`. However, when we receive status from the modem that we're disconnected, we are sent the status `NSAPI_STATUS_DISCONNECTED`. The connection is then later re-established by the modem, without us calling connect...

See `void read_reg_params_and_compare(RegistrationType type)` in `features/cellular/framework/AT/AT_CellularNetwork.cpp:149`.

What is really expected here? Should I get the status `NSAPI_STATUS_CONNECTING` instead, or should I expect the modem to re-attach also from `NSAPI_STATUS_DISCONNECTED`? Should I expect the state machine to return to `NSAPI_STATUS_DISCONNECTED` without calling the `NetworkInterface::disconnect()` method?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

AriParkkila, 2019-01-17:
@oscarh that's a good point for optimisation. `+CGEV` is subscribed to indicate that PDP context was really lost, but it seems that is not very well supported at the moment so we report disconnected also when registration is lost, e.g. `+CREG`. In the (near) future `CGEV` might be good to use..?

SeppoTakalo, 2019-01-17:
If the modem drops from the network, it should send status `NSAPI_STATUS_CONNECTING` and internally initiate reconnection logic.

NSAPI_STATUS_DISCONNECTED can only be produced as a result of `NetworkInterface::disconnect()` call.

If  there is difference, then the driver should be fixed.

SeppoTakalo, 2019-02-19:
Hi @oscarh 

After a internal review the current status is that:
1. Documentation is missing the arrow from "Connected" to "Disconnect" in case driver is not capable of handling the reconnection. We have decided that for Cellular, such promise does not always make sense, and for those case application should be able to decide whether the reconnection should happen. I'll update the documentation.
2. Some driver are still reworked to to return unified statuses, as they are documented.

closed 2019-02-19 by ciarmcom