ARMmbed/mbed-os #9383: NRF52840 BLE 5 MTU Size cant be bigger than 251

wlamprecht, 2019-01-15:
### Description

Hi 
I am trying to send packets(500 byte) from the NRFConnect Android app to the NRF52840 module.

Changes till now:

- Change RAM size
- Change NRF_SDH_BLE_GATT_MAX_MTU_SIZE to 513 (517 - L2CAP header size)
- Change NRF_SDH_BLE_GAP_EVENT_LENGTH to 5 
- Fix some data type bugs: uint8_t is not enough for MTU size 517

Changes in branch bug/MtuSize in repo evva-sfw
https://github.com/evva-sfw/mbed-os/tree/bug/MtuSize

Behaviour: 
If the packet sent from the android is bigger than 255, the nrfConnection app throws Error13(0xd): GATT INVALID ATTR LEN

As I know, the MTU size can be set up to 517 for BLE 5 devices. 
Is this a bug or is this not supported by the Nordic softdevice(S140)

Thanks, 
Werner 

Working with NRF52840 
Mbed 5.10 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

pan-, 2019-01-15:
@wlamprecht We are working on that issue in [this pr](https://github.com/paul-szczepanek-arm/mbed-os/pull/60/) (it is in a fork). Any feedback is welcome.

wlamprecht, 2019-01-16:
@pan- I think it is a different problem: 
mbed-os/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NRF52/source/btle/btle.cpp

```
        case BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST:
        {
            ble_gap_evt_t       const * p_gap_evt = &p_ble_evt->evt.gap_evt;

            uint8_t **const data_length_peer** =
                p_gap_evt->params.data_length_update_request.peer_params.max_tx_octets;

            const uint8_t **max_data_length** = NRF_SDH_BLE_GATT_MAX_MTU_SIZE + 4 /* L2CAP header size */;

            uint8_t const **data_length** = MIN(max_data_length, data_length_peer);
            ble_gap_data_length_params_t const dlp =
            {
                /* max_rx_octets */ data_length,
                /* max_tx_octets */ data_length
            };

            ASSERT_STATUS_RET_VOID(sd_ble_gap_data_length_update(p_gap_evt->conn_handle, &dlp, NULL));
        		break;
        }
```
First issue: The variables data_length_peer, max_data_length and data_length ar uint8_t therefore a MTU bigger than 255 is not possible.
Seccond issue: p_gap_evt->params.data_length_update_request.peer_params.max_tx_octets always returns 251 and the function MIN(max_data_length, data_length_peer) uses the min value from max_data_length and  data_length_peer therefore the MTU size cant be set to values bigger than 251 but is should be possible to set it to 517

pan-, 2019-01-16:
@wlamprecht Your issue will impact the implementation of the feature; I think these are related. 
@paul-szczepanek-arm Could you have a look ?

paul-szczepanek-arm, 2019-01-16:
Data length is the PDU size (packet size) and cannot be larger than 251 bytes.

That code should still be fixed to avoid truncation if MTU is larger than 251, thanks.

wlamprecht, 2019-01-17:
I fixed the mentioned bug on the fork https://github.com/evva-sfw/mbed-os/tree/bug/MtuSize
But the MTU can still not be changed to 517. 
I tried many different MTU sizes, but the BLE service does not work for MTU sizes bigger than 424. 
I tested everything with the NRF52840 chip, S140 soft device.
I tested many different settings, but it does not work

paul-szczepanek-arm, 2019-01-18:
What do you mean by "doesn't work"? Negotiates an ATT_MTU size smaller than you expect or negotiates the right size but then errors in some way? Crashes?

ATT_MTU size is agree on by both devices - are both set to the value you need?

The changes you have made are only relevant to over the air packet size. The BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST is used to neogtiate packet sizes with the peer. *NOT* ATT_MTU size. Attributes are transported broken up across multiple over the air packets. The "data length" is the size of those packets and has no relation to ATT_MTU size.

I am working on adding MTU events here:
https://github.com/paul-szczepanek-arm/mbed-os/pull/60

ATT_MTU size is exchanged in two places: one is BLE_GATTS_EVT_EXCHANGE_MTU_REQUEST below based on NRF_SDH_BLE_GATT_MAX_MTU_SIZE. You can make sure that the value there is what you think is being sent.

wlamprecht, 2019-01-18:
It has a strange behaviour, if I set BLE_GATT_MTU_SIZE_DEFAULT > 427. 

Scenario 1: 
Set BLE_GATT_MTU_SIZE_DEFAULT to 427 or lower.
Everything works as expected

Scenario 2: 
Set BLE_GATT_MTU_SIZE_DEFAULT to 428 or bigger.
In file mbed-os/features/FEATURE_BLE/ble/services/UARTService.h then the BLE_UART_SERVICE_MAX_DATA_LEN gets set to (BLE_GATT_MTU_SIZE_DEFAULT - 3). 
If (BLE_GATT_MTU_SIZE_DEFAULT - 3) is bigger tan 424, the RX characteristics disappears on the NRFConnect app(Android) . See attached pictures
[image]
[image]

paul-szczepanek-arm, 2019-02-18:
@wlamprecht this might be of interest to you:
https://github.com/ARMmbed/mbed-os/pull/9537

TacoGrandeTX, 2019-03-11:
@wlamprecht - did https://github.com/ARMmbed/mbed-os/pull/9537 help with this?

0xc0170, 2019-08-14:
> @wlamprecht - did #9537 help with this?

Closing as resolved via this PR.

closed 2019-08-14 by 0xc0170