ARMmbed/mbed-os #9390: Update icetea tests to include stderr output of failed command

cmonr, 2019-01-16:
### Description

When icetea fails pytest testing, it's error message is not useful.
An example: https://travis-ci.org/ARMmbed/mbed-os/jobs/480215039

Having debugged this for most of the day, it was found that the `icetea` command that was generating JSON to feed into the test was failing due to dependency resolution caused by a recent pyOCD release (0.15). 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

OPpuolitaival, 2019-01-23:
@cmonr it is a bit complex way how icetea is used but the message is there:

E           Exception: Error with python /home/travis/build/ARMmbed/mbed-os/tools/run_icetea.py --compile-list --mcu K64F --toolchain GCC_ARM --tcdir /home/travis/build/ARMmbed/mbed-os/tools/test/run_icetea/TEST_DIR_HW, 
E           return code: 1, 
E           error message: , 
E           output:Traceback (most recent call last):
E             File "/home/travis/build/ARMmbed/mbed-os/tools/run_icetea.py", line 284, in <module>
E               icetea_json_output = icetea_tests(options.target, options.tcdir, options.verbose)
E             File "/home/travis/build/ARMmbed/mbed-os/tools/run_icetea.py", line 177, in icetea_tests
E               list_json = json.loads(stdout)
E             File "/opt/python/2.7.14/lib/python2.7/json/__init__.py", line 339, in loads
E               return _default_decoder.decode(s)
E             File "/opt/python/2.7.14/lib/python2.7/json/decoder.py", line 364, in decode
E               obj, end = self.raw_decode(s, idx=_w(s, 0).end())
E             File "/opt/python/2.7.14/lib/python2.7/json/decoder.py", line 382, in raw_decode
E               raise ValueError("No JSON object could be decoded")
E           ValueError: No JSON object could be decoded
E           [ERROR] No JSON object could be decoded
tools/test/run_icetea/run_icetea_test.py:38: Exception

cmonr, 2019-01-30:
@OPpuolitaival Hmm, I guess so.

I suppose I was expecting stderr/stdout to also be printed, instead of needing to manually run the subcommand to find the root cause of the problem. Closing since I don't feel strongly about a change being needed.

closed 2019-01-30 by cmonr