ARMmbed/mbed-os #9396: Realtek RTL8195AM prints out extra messages on serial port

SeppoTakalo, 2019-01-16:
### Description

When running Mbed OS network tests+socket test RTL8195AM sometimes prints out extra messages into the serial port which may break the test result parsing.

For example, following message occasionally appear
```
[_rtx2_create_task] [rtw_little_wifi_mcu_thread]priority is higher than osPriorityRealtime
```

Also, following message appears on every boot:

```
gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003)

Check boot type form eFuse

SPI Initial

Image1 length: 0x4398, Image Addr: 0x10000bc8

Image1 Validate OK, Going jump to Image1

SDR Controller Init
```

These extra messages have possibility to break Mbed OS testing that depends on correct serial port output.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

SeppoTakalo, 2019-01-16:
CC: @ARMmbed/team-realtek please review.

0xc0170, 2019-01-16:
This does not fix it: https://github.com/ARMmbed/mbed-os/pull/9202 ?

SeppoTakalo, 2019-01-16:
Hmm..

It might have hidden all, but not boot messages.

Also the first example probably comes from RTX debug functions.

AdamZhang0124, 2019-01-18:
Hi @SeppoTakalo ,
Thank you for helping us discover this bug, I am debugging it now.

AdamZhang0124, 2019-01-21:
Hi @SeppoTakalo，

The ```[_rtx2_create_task] [rtw_little_wifi_mcu_thread]priority is higher than osPriorityRealtime``` warning message has been concealed.

However, as the following messages are burned in the ROM, they cannot be concealed.
```gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003)
Check boot type form eFuse
SPI Initial
Image1 length: 0x4398, Image Addr: 0x10000bc8
Image1 Validate OK, Going jump to Image1
SDR Controller Init 
```

Other than these, have you noticed any other redundant messages so far, so that I can solve them together with one commit? 

Many thanks.

samchuarm, 2019-02-14:
Hi @AdamZhang0124 , have you got the chance to check the `timer_wrapper` as Seppo suggested?

AdamZhang0124, 2019-02-19:
Hi @samchuarm , the `timer_wrapper` message seems to be a real error message rather than a debugging message. I have not encountered it before. As I checked, it is related to wlan, and if this message occurs, this error may cause the mismatch of the timer values.

Hi @SeppoTakalo , could you please help to double check if this error message still exists and whether this error would cause other problems?

Many thanks!

samchuarm, 2019-02-21:
Hi @SeppoTakalo , would you be able to help check if you can still see the `timer_wrapper` error message? Thanks.

samchuarm, 2019-03-11:
Hi @SeppoTakalo , can you please help check if this is still an issue or it can be closed? Thanks.

closed 2019-10-25 by 0xc0170