ARMmbed/mbed-os #9400: tests-mbedmicro-rtos-mbed-systimer to tune ?

jeromecoutant, 2019-01-16:
### Description

Hi

Since 5.11.1, STM32L0, STM32L4 and STM32F7 are TICKLESS.

Then tests-mbedmicro-rtos-mbed-systimer is now executed.

Test is OK for most of targets, except few one that are failing at test case 6:

````
[1547654364.30][CONN][RXD] >>> Running case #6: 'Handler called once'...
mbedgt: :252::FAIL: Values Not Within Delta 2500 Expected 42000 Was 45532
````

https://github.com/ARMmbed/mbed-os/blob/0b9587fe268b9414174a5b3752b209c54ebe590e/TESTS/mbedmicro-rtos-mbed/systimer/main.cpp#L252

I have the feeling that delay values should be tuned ?

Here is the failed tests I got with ST CI, including the difference value "t2 - t1", which is expected to be less than 2500:

Target | Toolchain | TestSuite | TestCase | result | Value
-- | -- | -- | -- | -- | --
DISCO_L072CZ_LRWAN1 | ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 3502
DISCO_L072CZ_LRWAN1 | GCC_ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 3593
DISCO_L072CZ_LRWAN1 | IAR | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 3929
DISCO_L476VG | GCC_ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 2555
DISCO_L496AG | GCC_ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 2677
NUCLEO_L073RZ | ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 3593
NUCLEO_L073RZ | GCC_ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 3654
NUCLEO_L073RZ | IAR | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 3807
NUCLEO_L476RG | GCC_ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 2525
NUCLEO_L486RG | GCC_ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 2556
NUCLEO_L496ZG | GCC_ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 2555
NUCLEO_L4R5ZI | ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 2586
NUCLEO_L4R5ZI | GCC_ARM | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 2800
NUCLEO_L4R5ZI | IAR | tests-mbedmicro-rtos-mbed-systimer | Handler called once | FAIL | 2525

@c1728p9 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2019-01-16:
@ARMmbed/mbed-os-hal Fyi

fkjagodzinski, 2019-01-17:
@jeromecoutant thanks for a detailed descripion. :+1: Could you run the updated code from #9416 on your targets, please? I could only test the `NUCLEO_L476RG` from the targets listed above; works OK now.