ARMmbed/mbed-os #9412: Cellular interface behaviour timeout change

vidavidorra, 2019-01-17:
### Description

Did something in the behaviour of the cellular interface timeouts change between the `OnboardCellularInterface` on Mbed OS 5.9.6 and `CellularBase` or `CellularContext` on Mbed OS 5.11.1?
I'm currently using the `CellularBase` as interface because that is supposed to be the most stable interface for cellular (see comments on #9270). When it gets a successful connection on the first try, I see the same behaviour as previously in that the connection is generally acquired in about 20 seconds. When, however, the connection is not successfully made on the fist try, it can now take around five minutes until the cellular interface gives up and returns an error code.
It looks like some new timeouts are implemented when I use the `CellularBase` or `CellularContext`, causing it to wait longer until it decides to fail on the connection. Is this the case and if so, how and where can I change this behaviour to match the timeouts/behaviour of the `OnboardCellularInterface`?

@AriParkkila @jarvte Maybe you guys know more about this as you seemed to know details about the cellular interface in #9270?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

vidavidorra, 2019-01-17:
To add: I didn't see any significant changes in the `mbed_config.h` regarding to AT timeout and cellular configuration.

AriParkkila, 2019-01-18:
@vidavidorra cellular 5.11 uses `CellularStateMachine` to connect. You can make its behavior similar to `PPPCellularInterface::connect` by setting connect retry count to 1. Because `set_retry_timeout_array()` is not exposed yet in 5.11, you need to call that directly from `CellularDevice` or modify `CellularStateMachine::_retry_timeout_array`.

vidavidorra, 2019-01-18:
@AriParkkila Thanks for your reply, will try that and post back.

In the last section of the failing connection on first try, see below, the section below is logged in about 20 seconds, but then hangs there for about a minute before retrying and succeeding in making the connection. Also see the log from successful connection on first try for comparison. Can you explain why it just seems to hang there for a minute when the connection on first try fails? It doesn't even try to resent the AT command or anything.

```
+CREG: 5,"00DE","F544"

+CGREG: 5,"00DE","F544"

+CGATT: 1

OK

OK
```

## Failing connection on first try
Connection takes e.g. from 2019-01-18 08:55:07 to 2019-01-18 08:56:25 (about 1 min 20 sec).

```
2019-01-18 08:55:07 [3] 0 requests loaded from storage (submodules/mbed-system/source/wtg_service/wtg_service.cpp:685)
[252]ATE0

OK

OK

+CME ERROR: 3

+CME ERROR: 3

+CPIN: READY

OK

OK

OK

OK

+CGREG: 2,2

OK

+CREG: 2,0

OK

+COPS: 0

OK

+CGREG: 2,2

OK

+CREG: 2,0

OK

+CGREG: 2,2

OK

+CREG: 2,0

OK

+CREG: 5,"00DE","F544"

+CGREG: 5,"00DE","F544"

+CGATT: 1

OK

OK
2019-01-18 08:56:19 [3] Failed to connect to network (0) (submodules/mbed-system/source/connection/connection.cpp:100)

+CME ERROR: 100
2019-01-18 08:56:19 [3] Failed to connect to network (0) (submodules/mbed-system/source/connection/connection.cpp:100)

OK

OK

CONNECT
2019-01-18 08:56:25 [3] Connected to network as IP_HIDDEN (submodules/mbed-system/source/connection/connection.cpp:98)
```

## Successful connection on first try
Connection takes e.g. from 2019-01-18 08:50:06 to 2019-01-18 08:50:25 (about 20 sec).

```
2019-01-18 08:50:06 [3] 3 requests loaded from storage (submodules/mbed-system/source/wtg_service/wtg_service.cpp:685)
[252]ATE0

OK

OK

+CME ERROR: 3

+CME ERROR: 3

+CPIN: READY

OK

OK

OK

OK

+CGREG: 2,2

OK

+CREG: 2,0

OK

+COPS: 0

OK

+CGREG: 2,2

OK

+CREG: 2,0

OK

+CGREG: 2,2

OK

+CREG: 2,0

OK

+CREG: 5,"00DE","F544"

+CGREG: 5,"00DE","F544"

+CGATT: 1

OK

OK

OK

CONNECT
2019-01-18 08:50:25 [3] Connected to network as IP_HIDDEN (submodules/mbed-system/source/connection/connection.cpp:98)
```

vidavidorra, 2019-01-18:
> You can make its behavior similar to PPPCellularInterface::connect by setting connect retry count to 1. Because set_retry_timeout_array() is not exposed yet in 5.11, you need to call that directly from CellularDevice or modify CellularStateMachine::_retry_timeout_array.

@AriParkkila I tried this by modifying `CellularStateMachine::_retry_timeout_array` and set it to 1. See ([#diff](#diff)). Doing that doesn't connect to the network at all. See the log below.
I'm currently trying the same with 2, which seems to be the lowest value for me when the connection is still being made.

```
2019-01-18 10:28:27 [3] 0 requests loaded from storage (submodules/mbed-system/source/wtg_service/wtg_service.cpp:685)
[252]ATE0

OK

OK

+CME ERROR: 3

+CME ERROR: 3

+CME ERROR: 3
2019-01-18 10:28:29 [3] Failed to connect to network (0) (submodules/mbed-system/source/connection/connection.cpp:100)
```

### Diff
```diff
diff --git a/features/cellular/framework/device/CellularStateMachine.cpp b/features/cellular/framework/device/CellularStateMachine.cpp
index 75e9416c1..0c6a0a102 100644
--- a/features/cellular/framework/device/CellularStateMachine.cpp
+++ b/features/cellular/framework/device/CellularStateMachine.cpp
@@ -60,14 +60,14 @@ CellularStateMachine::CellularStateMachine(CellularDevice &device, events::Event
     // set initial retry values in seconds
     _retry_timeout_array[0] = 1; // double time on each retry in order to keep network happy
-    _retry_timeout_array[1] = 2;
-    _retry_timeout_array[2] = 4;
-    _retry_timeout_array[3] = 8;
-    _retry_timeout_array[4] = 16;
-    _retry_timeout_array[5] = 32;
-    _retry_timeout_array[6] = 64;
-    _retry_timeout_array[7] = 128; // if around two minutes was not enough then let's wait much longer
-    _retry_timeout_array[8] = 600;
-    _retry_timeout_array[9] = TIMEOUT_NETWORK_MAX;
+//    _retry_timeout_array[1] = 2;
+//    _retry_timeout_array[2] = 4;
+//    _retry_timeout_array[3] = 8;
+//    _retry_timeout_array[4] = 16;
+//    _retry_timeout_array[5] = 32;
+//    _retry_timeout_array[6] = 64;
+//    _retry_timeout_array[7] = 128; // if around two minutes was not enough then let's wait much longer
+//    _retry_timeout_array[8] = 600;
+//    _retry_timeout_array[9] = TIMEOUT_NETWORK_MAX;
     _retry_array_length = RETRY_ARRAY_SIZE;
 }

diff --git a/features/cellular/framework/device/CellularStateMachine.h b/features/cellular/framework/device/CellularStateMachine.h
index afff6f05a..833a05684 100644
--- a/features/cellular/framework/device/CellularStateMachine.h
+++ b/features/cellular/framework/device/CellularStateMachine.h
@@ -32,7 +32,7 @@ class CellularPower;
 class CellularSIM;
 class CellularDevice;

-const int RETRY_ARRAY_SIZE = 10;
+const int RETRY_ARRAY_SIZE = 1;

 /** CellularStateMachine class
  *
```

AriParkkila, 2019-01-18:
@vidavidorra SIM probably needs more time to wake-up with `#define TIMEOUT_SIM_PIN (10*1000)` or you could loop connect repeatedly... `PPPCellularInterface::connect` is quite different from `CellularStateMachine` so you might want to copy connect logic from https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/cellular/generic_modem_driver/PPPCellularInterface.cpp

The log does not show AT commands (e.g. `AT+CPIN?`) but only responses (e.g. `+CPIN: READY`). Do you know which command was sent just before `+CME ERROR: 100`? It could be about finding and then creating a PDP context (but that should not take more than a few seconds).

vidavidorra, 2019-01-18:
@AriParkkila I've manually enabled the AT commands by enabling `ATHandler::debug_print` and `debug` in `mbed_debug.h`. I'll have a look at how to enable the commands, in addition to the responses. Unless you have some pointers to where I can enable those.
Hopefully the AT commands will give more insight in what is going on and why the interface seem to halt.

AriParkkila, 2019-01-18:
@vidavidorra to enable logs, define in `mbed_app.json`:
```
    "config": {
        "trace-level": "TRACE_LEVEL_DEBUG"
    },
    "target_overrides": {
        "*": {
            "mbed-trace.enable": true,
            "cellular.debug-at": true
        }
    }
```

vidavidorra, 2019-01-18:
I'm trying the same setup now but with both APN lookups disabled. See if that makes any difference.

AriParkkila, 2019-01-18:
@vidavidorra I don't see how `AT+CGDCONT=1,"IP","withthegrid.com"` can take over 60 seconds..? You could try to increase timeout `#define TIMEOUT_CONNECT (120*1000)` at `CellularStateMachine.cpp` to see if that makes any difference, and check if modem is responsive to AT commands during `AT+CGDCONT`.

vidavidorra, 2019-01-18:
> and check if modem is responsive to AT commands during AT+CGDCONT.

@AriParkkila How would I check this? I'm calling `_interface->connect();` where `_interface` is of type `CellularContext`. That call is blocking on my thread where I've created the interface, so I wouldn't be able to sent AT commands in the meantime. That is until `_interface->connect();` has returned, which it doesn't for at least 60 sec.
How do you propose I check modem responsiveness? Can I change something in the AT handler code where it sends `AT+CGDCONT` so I can try to send a command a couple of seconds after that? If so, could you point me to where I need to do so.

> You could try to increase timeout #define TIMEOUT_CONNECT (120*1000) at CellularStateMachine.cpp to see if that makes any difference

I'll try that until I hear back from you on the previous point.

> I'm trying the same setup now but with both APN lookups disabled. See if that makes any difference.

I've tested this and doesn't seem to make much difference.

vidavidorra, 2019-01-18:
@AriParkkila On this log, which is with APN lookup disabled, and the rest of the settings the same as the previous log. This log seems to hang at a different point and for less time (30 sec).

```
[WtgService::Connect] connection thread status: 8
2019-01-18 13:45:06 [3] 0 requests loaded from storage (submodules/mbed-system/source/wtg_service/wtg_service.cpp:685)
[252]ATE0
ATE0

OK
AT+CMEE=1

OK
AT+CGEREP=1

+CME ERROR: 3
AT+CPIN?

+CME ERROR: 3
AT+CPIN?

+CPIN: READY

OK
AT+CGREG=2

OK
AT+CREG=2

OK
AT+CGACT?

OK
AT+CGREG?

+CGREG: 2,2

OK
AT+CREG?

+CREG: 2,0

OK
AT+COPS?

+COPS: 0

OK
AT+CGREG?

+CGREG: 2,2

OK
AT+CREG?

+CREG: 2,0

OK

+CREG: 5,"0CB2","2FD4"
AT+CGATT?

+CGREG: 5,"0CB2","2FD4"

+CGATT: 1

OK
AT+CGDCONT?

OK
AT+CGDCONT=1,"IP","withthegrid.com"

OK
AT+CGDATA="PPP",1

CONNECT
2019-01-18 13:45:48 [3] Failed to connect to network (0) (submodules/mbed-system/source/connection/connection.cpp:100)
AT+CGDCONT?
2019-01-18 13:46:49 [3] Failed to connect to network (0) (submodules/mbed-system/source/connection/connection.cpp:100)
```

AriParkkila, 2019-01-18:
@vidavidorra that 30 seconds sounds like PPP connect failure at `AT_CellularContext::open_data_channel`. The reason could be `AT+CGDCONT?` command seen after `CONNECT` in your log.

That 60 second timeout is probably at `AT_CellularContext::set_new_context()`, where AT responsiveness could be checked:
```
    _at.set_timeout(1000); // first some small timeout
    _at.cmd_start("AT+CGDCONT=");
    _at.write_int(cid);
    _at.write_string(pdp_type_str);
    _at.write_string(_apn);
    _at.cmd_stop_read_resp();
    _at.sync(); // then send some commands to see modem is responsive
```

vidavidorra, 2019-01-21:
@AriParkkila It looks like the timeout you've noted makes a lot of difference. Previously I had about 20 % of the time where it wouldn't connect at all. I'm now running with Mbed trace propperly enabled and the changes from the attached patch (at timeout and at sync), and still have the other changes to decrease the retry array and increased `TIMEOUT_SIM_PIN`.
[cellular_context.patch.zip](https://github.com/ARMmbed/mbed-os/files/2779121/cellular_context.patch.zip).

Until now the new version (interface on Mbed OS 5.11.1) are performing perfect (36/36 connections made over almost 3 hours).
I'll backup all my changes and run the same test with only a timeout added before `_at.cmd_start("AT+CGDCONT=");` in the hope that that's what seemed to have fixed my issue.

Will report back with those results tomorrow hopefully. Can it make sense to you that it seems that the timeout before that command can be my issue? Why would a timeout be required there and why isn't it there in the source if it is, maybe specifically needed for my modem u-blox SARA-G350?

vidavidorra, 2019-01-22:
I've run the code with all the other changes resolved and only the timeout before `_at.cmd_start("AT+CGDCONT=");` and the AT sync after that. Unfortunately still running into the same issue. 137/187 connection made over roughly 15 hours.

I'll run some devices with different parts of the changes I made to get a more specific idea of what is required to get the same 100% score as I noted in my previous comment. Once I have that data I'll report back.

vidavidorra, 2019-01-22:
@AriParkkila I've been running a couple of test for the past hours.

1. Just debugging enabled.
    This has been running okay, over 90% of the connections made.
2. Debugging enabled and timeout before AT command.
    This has been running perfectly. 100% of the connections made within reasonable time.

My conclusion so far is that the fix for this is
1. Timeout before AT command
2. Debugging enabled, both the mbed_trace during connect and AT logs are needed. Only the mbed_trace still can't make some connections.
3. Maybe timeout after AT command and the `sync` after that.

Do you have any suggestion how to further debug this? Is there some generic timeout I can increase? It looks very much like the time it takes to send those debug messages to serial is a small delay/timeout that the program needs.

### Patch AT timeout + cmd
```diff
diff --git a/submodules/mbed-os/features/cellular/framework/AT/AT_CellularContext.cpp b/submodules/mbed-os/features/cellular/framework/AT/AT_CellularContext.cpp
index 1cb280318..f30e1322a 100644
--- a/submodules/mbed-os/features/cellular/framework/AT/AT_CellularContext.cpp
+++ b/submodules/mbed-os/features/cellular/framework/AT/AT_CellularContext.cpp
@@ -429,6 +429,7 @@ bool AT_CellularContext::set_new_context(int cid)

     //apn: "If the value is null or omitted, then the subscription value will be requested."
     bool success = false;
+    _at.set_at_timeout(1000); // first some small timeout
     _at.cmd_start("AT+CGDCONT=");
     _at.write_int(cid);
     _at.write_string(pdp_type);
@@ -436,6 +437,11 @@ bool AT_CellularContext::set_new_context(int cid)
     _at.cmd_stop_read_resp();
     success = (_at.get_last_error() == NSAPI_ERROR_OK);

+    printf("before AT try (timestamp: %lld)\n", time(nullptr));
+    _at.set_at_timeout(1000); // first some small timeout
+    _at.sync(AT_SYNC_TIMEOUT); // then send some commands to see modem is responsive
+    printf("after AT try (timestamp: %lld)\n", time(nullptr));
+
     // Fall back to ipv4
     if (!success && tmp_stack == IPV4V6_STACK) {
         _at.clear_error();
```

### Patch debug enable
```diff
diff --git a/submodules/mbed-os/platform/mbed_debug.h b/submodules/mbed-os/platform/mbed_debug.h
index f560f3db3..a727245ac 100644
--- a/submodules/mbed-os/platform/mbed_debug.h
+++ b/submodules/mbed-os/platform/mbed_debug.h
@@ -43,12 +43,12 @@ static inline void debug_if(int condition, const char *format, ...) MBED_PRINTF(
  */
 static inline void debug(const char *format, ...)
 {
-#if DEVICE_STDIO_MESSAGES && !defined(NDEBUG)
+//#if DEVICE_STDIO_MESSAGES && !defined(NDEBUG)
     va_list args;
     va_start(args, format);
     vfprintf(stderr, format, args);
     va_end(args);
-#endif
+//#endif
 }

@@ -62,14 +62,14 @@ static inline void debug(const char *format, ...)
  */
 static inline void debug_if(int condition, const char *format, ...)
 {
-#if DEVICE_STDIO_MESSAGES && !defined(NDEBUG)
+//#if DEVICE_STDIO_MESSAGES && !defined(NDEBUG)
     if (condition) {
         va_list args;
         va_start(args, format);
         vfprintf(stderr, format, args);
         va_end(args);
     }
-#endif
+//#endif
 }
```

AriParkkila, 2019-01-22:
@vidavidorra have you tried to add some delay between commands, e.g. `AT_CellularDevice::get_send_delay() {return 100;}`? Note that function may be overridden in your modem driver.

vidavidorra, 2019-01-22:
@AriParkkila No, I haven't tried that yet. It doesn't seem to be overridden in my case (u-blox PPP), so I've edited that function to return 200 for now and have disabled the traces to verify whether that works. I've also increased the `DEFAULT_AT_TIMEOUT` that is specified in `features/cellular/framework/AT/AT_CellularDevice.cpp`, to see whether that makes any difference. I'll test both combinations and report back.

vidavidorra, 2019-01-24:
@AriParkkila How can I set the `get_send_delay` from my application? Currently I've just edited the `AT_CellularDevice.cpp` source file in my Mbed OS fork but I'd prefer just configuring this from my application. The function is virtual, but is in my situation inherited by the `UBLOX_PPP` class. But that is directly used as far as I know, so it'd be hard for my application to inherit that and create an override function for `get_send_delay`. Do you have any advise on how to do this? Would you guys be open to exposing a function to set the delay that `get_send_delay` returns, I'm not sure yet on how this would work though?

The `AT+CGDCONT=` fix is a different story, I'm perfectly fine with having that fix in the Mbed OS source code.

vidavidorra, 2019-01-25:
@AriParkkila 
> `get_at_delay` is a timeout to wait for an AT response (not the same as `Thread::wait(1)`), and setting that might fix the problem due to it fails on the first call and then retry after a second succeeds..?

I don't know the exact internals, from what you've told me ([this comment](https://github.com/ARMmbed/mbed-os/issues/9412#issuecomment-456379631)) I thought that the `get_at_delay` is some timeout between commands and the `#define DEFAULT_AT_TIMEOUT 1000` (in `AT_CellularDevice.cpp`) is the response timeout. Which one is it?
<br>

> Do you think CellularDevice API should have setter for AT timeout (`set_send_delay`).

Yes! I'll happily make a PR for this.

What do you think about adding the `set_at_timeout` before the `AT+CGDCONT=`? Is that something that can be added in the Mbed source code via PR?

# Section

> Can you find any requirement like "send delay needed between AT commands" for your modem to define that in `UBLOX_PPP::get_send_delay()`?

I've looked in the datasheet of the SARA-G350 and the AT command manual, but couldn't find an explicit mention of such timeout. I did find the following:

- 20 ms wakeup time, but that would only be the case if the module is idle.
> When an OFF-to-ON transition occurs on the RTS input line, the UART is enabled and the module wakes up
to active mode: after ~20 ms from the OFF-to-ON transition the UART / module wake-up is completed and
data can be received without loss. The module cannot enter the low power idle mode and the UART is kept
enabled as long as the RTS input line is held in the ON state

- module active mode duration
> The module active mode duration outside an active call depends on:
 Network parameters, related to the time interval for the paging block reception (minimum of ~11 ms)
 Duration of UART enable time in absence of data reception (~20 ms)

https://www.u-blox.com/sites/default/files/SARA-G3-U2_SysIntegrManual_%28UBX-13000995%29.pdf#page=50
https://www.u-blox.com/sites/default/files/SARA-G3_DataSheet_%28UBX-13000993%29.pdf

AriParkkila, 2019-01-25:
@vidavidorra you are right. `_at.set_at_timeout(1000);` changes `DEFAULT_AT_TIMEOUT ` and `get_at_delay` is timeout between commands (or more precisely time from the previous response to the next command)... I guess `set_send_delay` API could be useful in general.

How _set_at_timeout before the AT+CGDCONT=?_ could fix the problem... Could it be that 1) Mbed sends `AT+CGDCONT=1,"IP","withthegrid.com"` too soon so modem dismisses the command, or 2) Mbed does not wait long enough for the response...?

vidavidorra, 2019-01-25:
@AriParkkila 
> `_at.set_at_timeout(1000);` changes `DEFAULT_AT_TIMEOUT` and `get_at_delay` is timeout between commands (or more precisely time from the previous response to the next command)...

Based on  that I'd expect that timeout before `AT+CGDCONT=?` shouldn't make any difference since the default timeout is 1000 ms, which is the same as what I've set it to, just before the command. I think I'll try to add a serial log just before the `AT+CGDCONT=?` is sent and print the `_at_timeout` at that moment. If it is already 1000, then adding the line before that wouldn't make any sense and shouldn't make any difference. If, however, the timeout is something different (< 1000) the issue is most likely somewhere else where the `DEFAULT_AT_TIMEOUT` is overwritten.

I'll test this, but that most likely won't happen today. I'll try and post back with results beginning of next week.

vidavidorra, 2019-02-05:
@AriParkkila I unfortunately haven't had time previously to continue with this.
I've just tried to print the timeout just before `_at.set_at_timeout(1000);` and that shows that the AT timeout is originally `60000` (would be `TIMEOUT_CONNECT` from the state machine). Below the values of `ATHandler::_previous_at_timeout`, `ATHandler::_at_timeout` and `ATHandler::_at_send_delay`, in that order.

```
Prev timeout: 60000, current: 60000, send delay: 100
After Prev timeout: 60000, current: 1000, send delay: 100
```

> How _set_at_timeout before the AT+CGDCONT=?_ could fix the problem...

After seeing this I agree with you. I was under the impression that this was an actual timeout, but now see that it simply changes the timeout value. I can't imagine how decreasing that could fix the problem. I can test again without changing the timeout. Has anything regarding the cellular interfaces etc. been changed between 5.11.1 and the latest release (5.11.3)? I'm on 5.11.1 and didn't see anything in the changelog that suggest changes that would affect this.
<br>

> 1) Mbed sends `AT+CGDCONT=1,"IP","withthegrid.com"` too soon so modem dismisses the command

How do you propose I test this?
<br>

> 2) Mbed does not wait long enough for the response...?

If this is the case, changing the timeout should make the problem worse, right?

vidavidorra, 2019-02-05:
Okay

1. If I remove the `_at.set_at_timeout(1000);` line, it starts to fail the connection again.
2. If I replace the `_at.set_at_timeout(1000);` with `wait_ms(100);` (imported from `mbed_wait_api.h` which is already included), it will succeed in the connection as 'normal'.
So seems that `_at.cmd_start("AT+CGDCONT=");` is sent too soon. @AriParkkila Could that be the case and how can I more accurately test that/change the `wait_ms` to a more specific value?

<br>
I also encountered another issue while testing.

When a connection is successfully made and the device does a disconnect after sending some data, the following connection fails (i.e. returns immediately after calling `connect()`) twice and creates a successful connection on the third call to `connect()`.
When I don't have the `wait_ms(100);` or `_at.set_at_timeout(1000);` it fails all calls to `connect()`. I'll print the returned code from connect to further analyse what is going on. I'm already handling `NSAPI_ERROR_IS_CONNECTED` there, but that is not the returned value.

AriParkkila, 2019-02-06:
@vidavidorra the delay before `AT+CGDCONT` (or between AT commands in general) should be defined in the manual of the modem, or need to ask from the manufacturer.

Is it possible that PPP connect takes more than 30 seconds? One reason that it succeeds every second time might be due to  `AT_CellularContext::disconnect()` seems to make also `AT+CGATT=0`, which probably should be removed to speed up connect. Can you remove it to see if it has any impact?

vidavidorra, 2019-02-06:
> the delay before `AT+CGDCONT` (or between AT commands in general) should be defined in the manual of the modem, or need to ask from the manufacturer.

I'll have a look in the u-blox AT commands/device manual to see whether anything specific is noted about this, but I doubt it.
<br>

> Is it possible that PPP connect takes more than 30 seconds?

I've seen the PPP connect to connect within a couple of seconds generally, so this seems very unlikely to me.
<br>

> One reason that it succeeds every second time might be due to `AT_CellularContext::disconnect()` seems to make also `AT+CGATT=0`, which probably should be removed to speed up connect. Can you remove it to see if it has any impact?

@AriParkkila You mean removing [this](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.11.1/features/cellular/framework/AT/AT_CellularContext.cpp#L684-L687) section?
I still don't understand how that would explain why the first 2 connect's don't succeed and the third does, consistently. I've tested this at least 10 times, and saw the same behaviour each time. Have you seen `0055.log` that I've attached to the previous comment? Specifically the `sema err:` prints that I'm doing in `AT_CellularContext::check_operation`. It counts from 3 and the connection succeeds when it reaches 1, hence the 2 connects that don't succeed. On the first run (line 140 in the log) it immediately prints 1, but on the 1st connect after the disconnect it is 3 (line 281, 297, 313). Do you know whether that can be related?

AriParkkila, 2019-02-06:
@vidavidorra yes, that code section should probably be removed completely.

It really seems in the log that semaphore release is not correct. Can you enable traces (or define all tr_ macros to printf):
```
    "config": {
        "trace-level": "TRACE_LEVEL_DEBUG"
    },
    "target_overrides": {
        "*": {
            "mbed-trace.enable": true,
            "cellular.debug-at": false
        }
    }
```

vidavidorra, 2019-02-06:
I currently seem to have a stable connection (also after the disconnect) on the first try when I changed the connect function to be async. I'm doing a `set_blocking(false)` before calling connect and `set_blocking(true)` after a connection has been made. Right after the connect call I'm going in a while loop and checking whether the required state (connected) has been achieved or my own timeout (40 s now) has been reached. So far, it has created the connection on the first try, even after the disconnect.

For now I'm continuing with the async connect as that seems to work for me. @AriParkkila Do you have any idea why the async call would work in this case?
Once I have time again, I can run the sync code with traces enabled to debug the semaphore thing. If the async works correctly it can be a while before I get to do that.

vidavidorra, 2019-02-06:
> > the delay before `AT+CGDCONT` (or between AT commands in general) should be defined in the manual of the modem, or need to ask from the manufacturer.
> 
> I'll have a look in the u-blox AT commands/device manual to see whether anything specific is noted about this, but I doubt it.

I've looked for this in the AT command manual and couldn't find a specific mention of a delay/time regarding this command. [u-blox UBX-13002752 - R60](https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf)

AriParkkila, 2019-02-06:
@vidavidorra async connect probably works because it does not use that (potentially incorrectly released) semaphore. We need to study this more, thanks for your findings.

Actually I'd say go with async whenever you can write your application that way :) You can listen for network callbacks (instead of 40 s wait) with something like:
```
static void network_callback(nsapi_event_t ev, intptr_t ptr)
{
    if (ev == NSAPI_EVENT_CONNECTION_STATUS_CHANGE && ptr == NSAPI_STATUS_GLOBAL_UP) {
             // indicate main to continue
    }
}

void main() {
    netif->set_blocking(false);
    netif->attach(callback(&network_callback));
    netif->connect();
    // wait until connect before creating sockets
}
```
You probably need to ask from u-blox for more details or use some experimental wait before `AT+CGDCONT`...

vidavidorra, 2019-02-06:
@AriParkkila Thanks for your clarification on the async functionality. I'll post my findings on sync with traces when I've got time for that, but as said, that probably will be a wile.

As for the callback, that is what I'm using, only I have added a timing around it to be able to stop trying to connect after a time I can set myself. During normal operation the idea is to continue based on the status value from the callback like you described!

AriParkkila, 2019-04-16:
@vidavidorra can we close this? The implementation of `CellularContext` in the latest Mbed OS with PR #10402 seems to be compatible with [NetworkInterface](https://os.mbed.com/docs/mbed-os/v5.12/apis/network-interfaces.html) passing  `mbed test -n tests-network-interface --app-config=tools/test_configs/CellularInterface.json`.

vidavidorra, 2019-04-17:
@AriParkkila Yeah, sorry. This can be closed now.

closed 2019-04-17 by vidavidorra