ARMmbed/mbed-os #9413: on-line IDE unable to load pull request

RobMeades, 2019-01-17:
### Description

Today we have been trying to import a Pull Request into the on-line IDE for review but, despite trying this from two different originating repos and into two different user accounts, we always get an error of the following form:

[image]

Note the "IDE" string in the URL rather than the usual "OS". Is there something up with the on-line IDE?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

RobMeades, 2019-01-17:
@fahim-ublox , @wajahat-ublox.

0xc0170, 2019-01-17:
I'll share this with the web team

cmonr, 2019-01-17:
I don't think this is a supported feature of the online compiler.

RobMeades, 2019-01-17:
It's what happens when you press the "Review" button on a Pull Request in the mbed dev team repos.

[image]

cmonr, 2019-01-17:
...huh. Good to know.

SamCHogg, 2019-02-01:
We pushed a fix for this today, can you please try again?

closed 2019-02-01 by ciarmcom

fahim-ublox, 2019-02-04:
Looks good to me