ARMmbed/mbed-os #9425: ESP8266: call MBED_ERROR if watchdog is triggered on modem

VeijoPesonen, 2019-01-18:
### Description
Driver should call MBED_ERROR instead of trying to recover from watchdog reset. This would make certain that modem issues wouldn't get hidden.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

VeijoPesonen, 2019-01-22:
Problem with MBED_ASSERT is that it's only effective with debug and development builds. I would saying failing as fast as possible is the best option here because we don't know the reason behind a watchdog reset and how often it might be occurring. If the device can't connect to a network it is better to see from the logs what is causing it in as early stage as possible.

SeppoTakalo, 2019-01-22:
MBED_ERROR causes device to reboot, and is therefore the correct action to get device back into known state.

VeijoPesonen, 2019-01-24:
PR #9456 implementing the change approved.