ARMmbed/mbed-os #9435: rtos::MemoryPool doesn't handle non-trivial (i.e. C compatible) types

Crzyrndm, 2019-01-18:
### Description
Background - I'm writing a desktop / portable implementation of the mbed rtos and utilities to enable more testing of low level infrastructure code (without the limitations imposed by working directly on the hardware). Due to this I've been digging quite deeply into some of the rtos constructs

Problem - rtos::MemoryPool is a simple wrapper over rtx_MemoryPool. This in itself is not an issue, but there is no protection in place preventing (or code to allow) a user using rtos::MemoryPool with a type (T) that is not C compatible (e.g. a smart pointer member which relies on it's destructor being called to free linked memory).

I see this issue (https://github.com/ARMmbed/mbed-os/issues/5891) which pointed out the lack of documentation around this, but I have to wonder why not just fix the issue at source?

Affects
* rtos::MemoryPool
* rtos::Mail (based on MemoryPool)

This is relatively easy to resolve in my project by wrapping it in a class which calls ctor/dtor on the way in and out, but it seems like a footgun in the library which should be resolved (either through asserting that the type is compatible with the current implementation, or adding ctor/dtor calls).

The assert is dead easy with C++11 compilers but could be a little more difficult if that is not available
```
#if __cplusplus >= 201103L
#include <type_traits>
#endif

#if __cplusplus >= 201103L
static_assert(std::is_pod<T>::value, "rtos::MemoryPool can only handle C compatible types");
#endif
```
Placement new/delete is I believe standard right through from C++98 and doesn't require any includes
```
// alloc
return new (osMemoryPoolAlloc(_id, 0)) T(); // run default ctor on block of memory returned by the memory pool. The "new" here doesn't allocate at all
// free
block->~T(); // manually call dtor of T. No deallocation of memory
return osMemoryPoolFree(_id, (void *)block);
```

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2019-01-21:
> Background - I'm writing a desktop / portable implementation of the mbed rtos and utilities to enable more testing of low level infrastructure code (without the limitations imposed by working directly on the hardware). Due to this I've been digging quite deeply into some of the rtos constructs

I am eager to hear more about this one, will you be able to share more?

> I see this issue (#5891) which pointed out the lack of documentation around this, but I have to wonder why not just fix the issue at source? Having ctor/dtor is one of the biggest advantages of working in C++ instead of C and the fact they are missing here is very non-obvious.

@ARMmbed/mbed-os-core Please review.

kjbracey-arm, 2019-01-22:
Makes sense that `MemoryPool` should be able to do this, but changing `alloc` and `free` would not be a backwards-compatible change - anyone already using it with C++ may well be doing their own placement new in the pointer returned, and that requirement is documented.

You would have to make it new methods, rather than changing `alloc` and `free`.

Feels like it's a teeny bit backward though - why would any user of your class be explicitly calling the `MemoryPool`?

If you have a system pool for a given object, should that object not have its own `new` and `delete` methods to force allocation via that pool, transparently to the user of the class? And they would want to call the "plain" alloc and free from the pool, and then normal constructors and destructors would be called.

kjbracey-arm, 2019-01-23:
I guess my point is that in C++ code should not usually be explicitly calling "alloc" or "free" functions to allocate or delete objects, just as they shouldn't be calling `::malloc` or `::free`. They should really be using `new` and `delete`, or an Allocator.

If working that way, the implementation of the custom new/delete operators, or the Allocator, will be using the plain alloc/free from MemoryPool, but users of the object won't be seeing that detail.

Certainly we could add new MemoryPool methods, but that is the conceptual equivalent of giving you 

    template<class T>
    T *malloc();

and expecting applications to do `Foo *object = malloc<Foo>()` instead of `Foo *object = new Foo()`. That's just not idiomatic C++.

I tend to agree that the existing `MemoryPool::alloc` method is misleading - being a raw allocator, suitable for making a custom `new` or Allocator, it should arguably be returning `void *`, not `T *`.  Although I note that a C++ Allocator's `allocate` does actually return a pointer, not `void *`, despite not constructing, so this is consistent with that.

It seems that `MemoryPool` could be upgraded to a standard C++ Allocator, which would give it some value.

You could write your code like this:

    template<class T>
    void *operator new(std::size_t count, MemoryPool<T> &pool)
    {
        assert(count == sizeof(T));
        return pool.alloc();
    }

    MemoryPool<Obj> pool;
    Obj *object = new (pool) Obj()

Maybe `MemoryPool` should provide that `new` overload itself.

Or if you're defining the type `Obj` yourself, you could include your own custom `new` which means it automatically gets allocated from the pool whenever anyone uses `new Obj`.

Crzyrndm, 2019-01-23:
Fair call on not using *alloc in a C++ program without some care (and I like the idea of the new overload for the pool, hadn't thought of that). Many of the firmware devs I know are more C-based though so it doesn't have quite the same red flags for everyone (have had reactions that would result in errors ranging from assuming the ctor was run, to mixing C style memset init with a C++ class).

RE: std::allocator/allocator_traits returning a T*, I don't neccesarily disagree, but will point out that they additionally provide construct/destroy (a strong hint that the 4 operations are seperate). 

I think this issue has served it's purpose from my end in prompting some conversation.
I will be experimenting with a more invasive (i.e. not drop-in) pool allocator interface to see if I can find a middle-ground that is less error-prone and still explicit enough about the memory pool usage (I'm not sure if the "new with allocator" will be as cleanly understood as a function call since it's not seen very often)