ARMmbed/mbed-os #9450: Wio 3G (UG96) always failed to run at the 2nd connection

toyowata, 2019-01-22:
### Description

2nd time cellular connection of the Wio 3G (UG96) always failed (TIMEOUT) for cellular and Pelion-ready example code.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

## Step to reproduce
```
$ mbed import mbed-os-example-cellular
$ cd mbed-os-example-cellular
$ mbed ls
[mbed] Working path "/Users/toywat01/test/seeed/wio/mbed-os-example-cellular" (program)
mbed-os-example-cellular (#42237627b75a, tag: mbed-os-5.11.2)
`- mbed-os (#a8f0c33eaa2c, tags: latest, mbed-os-5.11.2)

```
* Set correct APN/User/Password information in the `mbed_app.json` and add following config as an workaround of the issue #9196.
 ```json
        "WIO_3G": {
            "target.device_has_remove" : ["SERIAL_FC"]
        }
```
* Build the example
```
$ mbed compile -m wio_3g -t gcc_arm
```
* Flash the example and press reset button to run

[1st runtime log output - works fine]
```
-example-cellular
Establishing connection
....
Connection Established.
TCP: connected with echo.mbedcloudtesting.com server
TCP: Sent 4 Bytes to echo.mbedcloudtesting.com
Received from echo server 4 Bytes
.
Success. Exiting 
```
[2nd runtime log output - TIMEOUT]
(press reset button here)
```
-example-cellular
Establishing connection
.................................................................
```
This will be timeout error, finally.  It sometimes (less than 50% time) worked as expected, but faild in most of 2nd case.

## simple-mbed-cloud-client-tests-dev_mgmt-connect test

Test result of the simple-mbed-cloud-client-tests-dev_mgmt-connect test case: 
[SMCC-dev_mgmt-connect-wio_3g-failed.txt](https://github.com/ARMmbed/mbed-os/files/2781619/SMCC-dev_mgmt-connect-wio_3g-failed.txt)

It always worked fine at the 1st connection `{{iteration;0}}`, but failed at 2nd connection `{{iteration;1}}`.  Below is log output with "cellular.debug-at” enabled.

[1st connection log]
```
[1548140190.73][SERI][TXD] {{iteration;0}}
[1548140190.74][CONN][RXD]
[1548140190.74][CONN][INF] found KV pair in stream: {{__testcase_name;Resource LwM2M POST}}, queued...
[1548140190.75][CONN][RXD] >>> Running case #1: 'Connect to Cellular'...
[1548140190.75][CONN][INF] found KV pair in stream: {{__testcase_start;Connect to Cellular}}, queued...
[1548140190.76][CONN][RXD] [INFO] Attempting to connect to network.
[1548140190.77][CONN][RXD] AT
[1548140191.78][CONN][RXD] AT
[1548140192.57][CONN][RXD] ATE0
[1548140195.58][CONN][RXD] ATE0
[1548140200.59][CONN][RXD] ATE0
[1548140200.60][CONN][RXD] atE0
[1548140200.60][CONN][RXD]
[1548140200.61][CONN][RXD] OK
[1548140200.61][CONN][RXD] AT+CMEE=1
[1548140200.62][CONN][RXD]
[1548140200.64][CONN][RXD] OK
[1548140200.64][CONN][RXD] AT+CGEREP=1
[1548140200.66][CONN][RXD]
[1548140200.66][CONN][RXD] OK
[1548140200.66][CONN][RXD] AT+CPIN?
[1548140200.68][CONN][RXD]
[1548140200.69][CONN][RXD] +CPIN: READY
[1548140200.69][CONN][RXD]
[1548140200.70][CONN][RXD] OK
[1548140200.71][CONN][RXD] AT+CGREG=2
[1548140200.73][CONN][RXD]
[1548140200.73][CONN][RXD] OK
[1548140200.73][CONN][RXD] AT+CREG=2
[1548140200.75][CONN][RXD]
[1548140200.75][CONN][RXD] OK
[1548140200.75][CONN][RXD] AT+CGACT?
[1548140200.77][CONN][RXD]
[1548140200.77][CONN][RXD] OK
[1548140200.78][CONN][RXD] AT+CGREG?
[1548140200.80][CONN][RXD]
[1548140200.81][CONN][RXD] +CGREG: 2,2
[1548140200.82][CONN][RXD]
[1548140200.82][CONN][RXD] OK
[1548140200.82][CONN][RXD] AT+CREG?
[1548140200.83][CONN][RXD]
[1548140200.85][CONN][RXD] +CREG: 2,2
[1548140200.85][CONN][RXD]
[1548140200.85][CONN][RXD] OK
[1548140200.85][CONN][RXD] AT+COPS?
[1548140200.85][CONN][RXD]
[1548140200.88][CONN][RXD] +QIND: PB DONE
[1548140200.88][CONN][RXD]
[1548140200.89][CONN][RXD] +COPS: 0
[1548140200.89][CONN][RXD]
[1548140200.89][CONN][RXD] OK
[1548140202.90][CONN][RXD] AT+CGREG?
[1548140202.92][CONN][RXD]
[1548140202.93][CONN][RXD] +CGREG: 2,2
[1548140202.93][CONN][RXD]
[1548140202.94][CONN][RXD] OK
[1548140202.94][CONN][RXD] AT+CREG?
[1548140202.95][CONN][RXD]
[1548140202.96][CONN][RXD] +CREG: 2,2
[1548140202.98][CONN][RXD]
[1548140202.98][CONN][RXD] OK
[1548140203.73][CONN][RXD]
[1548140203.75][CONN][RXD] +CGEV: NW CLASS A
[1548140203.75][CONN][RXD]
[1548140203.76][CONN][RXD] +CREG: 2
[1548140203.76][CONN][RXD]
[1548140203.79][CONN][RXD] +CGREG: 1,"00AC","468C002",2
[1548140203.81][CONN][RXD] AT+CGATT?
[1548140203.82][CONN][RXD]
[1548140203.83][CONN][RXD] +CGATT: 1
[1548140203.83][CONN][RXD]
[1548140203.84][CONN][RXD] OK
[1548140203.84][CONN][RXD] AT+CGDCONT?
[1548140203.86][CONN][RXD]
[1548140203.86][CONN][RXD] OK
[1548140203.86][CONN][RXD] AT+CGDCONT=1,"IP","soracom.io"
[1548140203.91][CONN][RXD]
[1548140203.91][CONN][RXD] OK
[1548140203.92][CONN][RXD] AT+CGDATA="PPP",1
[1548140203.94][CONN][RXD]
[1548140210.32][CONN][RXD] CONNECT[INFO] Connected to network successfully. IP address: 10.129.102.246
```

[2nd connection log]
```
[1548140248.03][CONN][RXD] >>> Running case #1: 'Connect to Cellular'...
[1548140248.03][CONN][INF] found KV pair in stream: {{__testcase_start;Connect to Cellular}}, queued...
[1548140248.03][SERI][TXD] {{iteration;1}}
[1548140248.04][CONN][RXD] [INFO] Attempting to connect to network.
[1548140248.04][SERI][TXD] {{iteration;1}}
[1548140248.05][CONN][RXD] AT
[1548140248.05][SERI][TXD] {{iteration;1}}
[1548140248.07][CONN][RXD]
[1548140248.09][CONN][RXD] OK
[1548140248.09][CONN][RXD] AT
[1548140248.10][CONN][RXD]
[1548140248.11][CONN][RXD] OK
[1548140248.11][CONN][RXD] ATE0
[1548140248.13][CONN][RXD]
[1548140248.13][CONN][RXD] OK
[1548140248.14][CONN][RXD] AT+CMEE=1
[1548140248.17][CONN][RXD]
[1548140248.17][CONN][RXD] OK
[1548140248.18][CONN][RXD] AT+CGEREP=1
[1548140248.20][CONN][RXD]
[1548140248.21][CONN][RXD] OK
[1548140248.21][CONN][RXD] AT+CPIN?
[1548140248.23][CONN][RXD]
[1548140248.25][CONN][RXD] +CPIN: READY
[1548140248.25][CONN][RXD]
[1548140248.25][CONN][RXD] OK
[1548140248.26][CONN][RXD] AT+CGREG=2
[1548140248.29][CONN][RXD]
[1548140248.30][CONN][RXD] OK
[1548140248.30][CONN][RXD] AT+CREG=2
[1548140248.32][CONN][RXD]
[1548140248.32][CONN][RXD] OK
[1548140248.33][CONN][RXD] AT+CGACT?
[1548140248.35][CONN][RXD]
[1548140248.38][CONN][RXD] +CGACT: 1,1
[1548140248.38][CONN][RXD]
[1548140248.38][CONN][RXD] OK
[1548140248.38][CONN][RXD] AT+CGREG?
[1548140248.41][CONN][RXD]
[1548140248.44][CONN][RXD] +CGREG: 2,1,"00AC","4680002",2
[1548140248.44][CONN][RXD]
[1548140248.44][CONN][RXD] OK
[1548140248.44][CONN][RXD] AT+CGATT?
[1548140248.48][CONN][RXD]
[1548140248.49][CONN][RXD] +CGATT: 1
[1548140248.49][CONN][RXD]
[1548140248.50][CONN][RXD] OK
[1548140248.50][CONN][RXD] AT+CGDCONT?
[1548140248.52][CONN][RXD]
[1548140248.58][CONN][RXD] +CGDCONT: 1,"IP","soracom.io","10.129.102.246",0,0
[1548140248.58][CONN][RXD]
[1548140248.59][CONN][RXD] OK
[1548140248.59][CONN][RXD] AT+CGDATA="PPP",1
[1548140248.62][CONN][RXD]
[1548140248.64][CONN][RXD] +CME ERROR: 148
[1548140248.67][CONN][RXD] [WARN] Unable to connect to network. Retrying...AT+CGDCONT?
[1548140248.70][CONN][RXD]
[1548140248.75][CONN][RXD] +CGDCONT: 1,"IP","soracom.io","10.129.102.246",0,0
[1548140248.75][CONN][RXD]
[1548140248.77][CONN][RXD] OK
[1548140248.77][CONN][RXD] AT+CGDATA="PPP",1
[1548140248.80][CONN][RXD]
[1548140248.82][CONN][RXD] +CME ERROR: 148
[1548140248.84][CONN][RXD] [WARN] Unable to connect to network. Retrying...AT+CGDCONT?
[1548140248.87][CONN][RXD]
[1548140248.93][CONN][RXD] +CGDCONT: 1,"IP","soracom.io","10.129.102.246",0,0
[1548140248.93][CONN][RXD]
[1548140248.93][CONN][RXD] OK
[1548140248.94][CONN][RXD] AT+CGDATA="PPP",1
[1548140248.97][CONN][RXD]
[1548140248.99][CONN][RXD] +CME ERROR: 148
[1548140249.03][CONN][RXD] [WARN] Unable to connect to network. Retrying...[ERROR] Device failed to connect to Network.
[1548140249.04][CONN][INF] found KV pair in stream: {{test_failed;1}}, queued...
[1548140249.04][CONN][INF] found KV pair in stream: {{__testcase_finish;Connect to Cellular;1;1}}, queued...
[1548140249.04][HTST][INF] __notify_complete(False)
[1548140249.04][HTST][INF] __exit_event_queue received
[1548140249.04][HTST][INF] test suite run finished after 58.58 sec...
[1548140249.05][CONN][RXD] >>> 'Connect to Cellular': 1 passed, 1 failed
```

cc @ARMmbed/team-seeed @ARMmbed/mbed-os-wan

AriParkkila, 2019-01-22:
@toyowata can you check that `NetworkInterface::disconnect()` is called before exiting the testcase.

toyowata, 2019-01-22:
@AriParkkila 

>can you check that `NetworkInterface::disconnect()` is called before exiting the testcase.

I actually did it and not working.

I am discussing this issue with @ARMmbed/team-seeed and following is possible fix (force power off the modem in startup).

* Implement `onboard_modem_power_down()` function for UG96
```c++
void onboard_modem_power_down()
{
    gpio_t gpio;
    /* keep the power line low for 1 second */
    //press_power_button(1000);

    gpio_init_out_ex(&gpio, M_POWR, 0);
    wait_ms(1000);
}
```
* Force power off the modem module at startup
```c++
QUECTEL_UG96_CellularPower::QUECTEL_UG96_CellularPower(ATHandler &atHandler) : AT_CellularPower(atHandler)
{
    // Force power off the modem here, 
    // so it is always initialized to the STATE_POWER_ON state
    // in the CellularStateMachine::state_init()
    off();
}
```

toyowata, 2019-03-05:
Closed by #9554 and #9555

closed 2019-03-05 by toyowata