ARMmbed/mbed-os #9451: USBSerial baud rate setting does not work

maskedw, 2019-01-22:
### Description
+ Target: NUCLEO_L476RG
+ Toolchain: ARM-GCC 6.3.1
+ Mbed: mbed-os-5.10.4

I am developing applications using USBSerial.

`mbed-os/features/unsupported/USBDevice/USBSerial`

NUCLEO_L476RG was not supported by the standard, but it was able to operate USB with a little fix.

`mbed-os/features/unsupported/USBDevice/targets/TARGET_STM`

Data send / receive is working normally.But the transmission speed is too fast.

+ Setting PC Terminal Application: baud 115200baud
+ 64 KByte data transmission (PC => Mbed)
+ Transfer time is about 100ms (Aboud 8.5Mbaud)

How do PC applications recognize the baud rate setting?
It seems that it is set to 9600 baud for line_condig. Is this setting ignored or data not received?
```
// `mbed-os/features/unsupported/USBDevice/USBSerial/USBCDC.cpp`
// 9600baud setting ?
static uint8_t cdc_line_coding[7]= {0x80, 0x25, 0x00, 0x00, 0x00, 0x00, 0x08};
```

### Issue request type

    [x] Question
    [ ] Enhancement
    [] Bug

kjbracey-arm, 2019-01-22:
The host can request a serial speed by sending "SET_LINE_CODING".

But this is a virtual serial link - the emulation does not extend to emulation of slow serial speeds. The speed will be as fast as is possible via USB.

The line config setting would be used if the device was a USB->serial converter, or a modem, to configure the speed of the physical serial port. It appears that the data there is just dummy data to ensure we have something to report if asked for line coding, presumably for compatibility with some hosts. The device is ignoring set requests. (In principle, the device shouldn't have to support get/set line coding at all, as it's not a converter).

If you did want to make a converter application, then you would want an extra API to allow the to application read the line coding set by the host, and program that into the real serial port. The data rate would then be flow controlled by that real serial port.

maskedw, 2019-01-23:
USBDevice (Mbed) performs speed adjustment according to the baud rate setting.
Does USBHost (PC) do speed adjustment?

kjbracey-arm, 2019-01-23:
I don't quite understand the question. The USB CDC link provides two ways for speed to be indicated:

* The Host can issue SET_LINE_CODING to request a connection speed change from the Device
* The Device can report CONNECTION_SPEED_CHANGE to indicate a speed change on its connection

The first is definitely intended for UART compatibility, the second is more for Ethernet or similar

But the USB CDC link itself does not use those speeds - it runs at full USB rate.    If you were using USBSerial to make a protocol converter, then you should be changing the speed or reporting the speed of your serial interface or ethernet interface with those USB commands, but that wouldn't affect the speed of the USB link itself, at least not directly.

If you really need your communications to be slow, then either:

* you will need some sort of throttling code on the PC to force it to send data slowly over a virtual UART, according to its nominal baud setting - I did a search to see if such a utility already exists, but can't see anything. Only people asking how to go faster...

* you should write your Mbed device application to look at the speed received via "SET_LINE_CODING" (that will need to be exposed via API on USBSerial), and make it artificially limit data reading and writing rate to that using a timer.

maskedw, 2019-01-27:
I realized that both the USB Host and the USBSerial class of mbed do not consider the baud rate setting.

closed 2019-01-27 by maskedw